MEGA65/mega65-core #598: MOUNT D81 image + RESET via FREEZER fails

MauriceMEGA65, 2022-07-29:
**Test Environment (required)**
Core  3253c5d, including Essential files (https://discord.com/channels/719326990221574164/791383472853614593/998239363722911854)
ROM 920376

**Describe the bug**
MOUNT+RESET via FREEZER -> Fail. Reset via Freezer (F5) boots the previous mounted image, not the just selected one
https://discord.com/channels/719326990221574164/791383472853614593/998931384804573225

**To Reproduce**
Mounting an image via BASIC (use 'MOUNT') works fine. Reset the machine and do DIR will show the new mounted image.
Mounting an image via Freezer, then select F5 to reboot, then do a DIR will show not the selected image, but the previously mounted one.

**Expected behavior**
Same like mounting via BASIC; after a reset and do DIR, the newly mounted D81 image should be visible

**Additional context**
Interestingly, using the new Freezer on the release core, shows not this issue.
I have done some experiments with old (=R3A machines) Essential files/core/ROM and the new set (link above).
My findings indicates that in whatever mixed configuration I'm in, only the configurations that includes the 3253c5d core fails.
So, it looks like a core issue to me, not a Freezer or ROM issue.

ps: not being able to do a proper mount+reset might also be the reason for failing to load GEOS on the new core.

lydon42, 2022-09-26:
Can't confirm with RC1 please recheck.

lydon42, 2022-10-02:
I would rule this was a bug and was fixed by the recent freezer/hyppo enhancements regarding image handling.
Can't reproduce on RC1.

closed 2022-10-02 by lydon42