MEGA65/mega65-core #600: High-quality high-speed random number generator

gardners, 2022-08-08:
**Is your feature request related to a problem? Please describe.**
The MEGA65 currently lacks a high-speed good-quality hardware random number generator.  Having one would make a lot of software easier to write, for example to simulate the behaviour of various productivity software.

**Describe the solution you'd like**
A free IO register should return a fresh random value whenever read.

**Describe alternatives you've considered**
Doing nothing. Always a noble option.

**Additional context**
None

gardners, 2022-08-08:
Use neoTRNG

lydon42, 2023-05-09:
Problem is the toggle of trng_consume_toggle happening in function read_data_complex: you can not change signals within functions.
Workaround using write to RNG register for toggelung results in random bytes read.
Trying different method of toggling on read outside of the function next.

lydon42, 2023-05-10:
# Testing the RNG
It seems that the RNG takes a lot of cycles to produce a new number. The picture shows the number changing every 16-22 bytes. The loop takes 16 cycles, which translates to 256 to 352 cycles for a new number.
This means to me that we need access to the valid bit of neoTRNG so that software can determine if the RNG already has a new number.
[image]

closed 2024-02-24 by lydon42