MEGA65/mega65-core #601: C65 demo does not run on core 3253c5d, but does run on XEMU and slot 0 core

MauriceMEGA65, 2022-08-22:
**Test Environment (required)**
You can use [MEGA65INFO](https://files.mega65.org?id=3eeb7d82-8bd7-44b7-82b6-cef29d123b0e) to retrieve this.
 - Platform: MEGA65
 - Core Commit: 3253c5d
 - ROM Release: 911001

**Describe the bug**
C65 demo (see attached) does not run on latest core, but runs OK on core in slot 0 and on XEMU

**To Reproduce**
Steps to reproduce the behavior:
1. Boot with old ROM (911001)
2. Mount ZED-YAGO.D81 (see attached)
[zed_yago.d81.zip](https://github.com/MEGA65/mega65-core/files/9397446/zed_yago.d81.zip)

3. Load and run DEMOLOAD65 (will switch to C64 mode) 
4. In C64 mode, load and run DEMOLOAD64

**Expected behavior**
Same as on XEMU or real hardware booted with slot 0 core

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

lgblgblgb, 2022-08-22:
I'm not sure how important that is, since: https://github.com/lgblgblgb/xemu/issues/142#issuecomment-855451262

So it can happen it's just a coincidence that it runs with older bitstream (slot 0) or Xemu. But I can't judge this clearly anyway.

lydon42, 2022-10-02:
A slot 0 core is just a very old core. As we are adding enhancements, old stuff intended for the "real" C65 will probably break, it is not possible to enhance and keep old the old "quirks".
So if you like to run a very old example, make sure to use an equally old core.

closed 2022-10-02 by lydon42