MEGA65/mega65-core #622: Make activating the floppy drive to make sounds while accessing SD, an option

MauriceMEGA65, 2022-10-02:
Currently the floppy drive is being activated while accessing the SD in order to create floppy drive sounds.

I would like to be able to disable the floppy drive being activated while accessing the SD. Using both media makes it hard to distinguish which drive is being read/written by the sound of it. An enable/disable setting in config would be welcome. Perhaps also reducing wear and tear as a bonus. I will make it a future request

lydon42, 2022-10-02:
Linked to #621

MauriceMEGA65, 2022-10-02:
Ah, same issue as #518, so most likely duplicate

grubi, 2023-01-09:
I'd like to support this one. Drive sounds are no problem when loading a game from time to time. But when the Mega65 is used for heavy development, it can quickly become annoying - plus I fear it will wear off the old, used drives more quickly. I have disconnected my disk drive cable internally, after finding out that the SILENT flag in $D6A0 doesn't work.

tmilker, 2023-01-26:
I'd also like to express my support for this feature. In addition to not being able to determine if a real disk is being accessed, what happens if we have an actual disk in the drive?  I don't know really know how a floppy drive works, could this cause wear & tear on the disk itself too or an unknown/unexpected bug cause it to be accidentally written to or damaged?

KiDraDragon, 2023-01-26:
It would be a very much appreciated feature.

Mikey351AU, 2023-01-26:
I support this request.

LeifBloomquist, 2023-01-27:
Yes!  Need this

nobruinfo, 2023-02-05:
I would even suggest to make this *opt-in*. Meaning it should not only be configurable but **off** at default. Not only would this make it more VICE compatible :) but also people would not have surprise effects upon first starts of their machines.

lydon42, 2023-02-18:
This needs #621 to be fixed.

deathybrs, 2023-08-10:
I add my voice to this request - it is a really cool feature, but many of us fear for the life of our drives, which seems like a valid concern to me.  I further really like @nobruinfo's suggestion that the option default to off rather than on so that we start out protecting people's disk drive.

dansanderson, 2023-09-17:
Notes from Paul:

Most likely in sdcardio.vhdl
the line that enables the real drive motor is motor <= fastio_wdata(5)
hang on, no, it will be f_motora/f_motorb
silent_sdcard is the variable that controls it, and it should be working at first look.
It is not currently applied to head stepping, however.
so this is likely what is happening: The actual selection of real drive is suppressed until you step tracks, at which point it does get selected.
basically follow the f_selecta/b references and add the missing checks for real floppy or silent_sdcard to those blocks.

lydon42, 2023-09-18:
If unsure: also add to ONBOARD, so the user gets asked as soon as possible.

RetroCogs, 2024-01-09:
As someone that develops late at night, the drive sounds everytime I launch an iteration are very loud and I would love to be able to develop in silence.

tundraC, 2024-01-11:
I'd love the option to en/disable.  I was troubleshooting a boot issue last year and it was very difficult to figure out what was being used for loading.  The issue was very hard to determine if the drive was actually seeking or faking it.  Plus, when working late at night, it's nice to keep it quiet.  Just my two thoughts :)

dansanderson, 2024-07-02:
I can confirm that this is implemented and working in the latest development core. This issue can be closed (along with #621).

closed 2024-09-10 by ki-bo

MauriceMEGA65, 2024-09-10:
👌