MEGA65/mega65-core #625: MEGAFLASH: add JTAG-less slot 0 flashing option

lydon42, 2022-10-04:
If slot 0 is erased, a slot 1 core will still be booted after 30 seconds of searching. This also happens if only the start of slot 0 is erased, or some piece inside slot 0. My tests always ended up booting slot 1 after a short while.

So we could make a slot 0 flasher by requiring a good core in slot 1 and asking the user if he is able to start the system from slot 1.

This would enable us to add the Slot 0 flash function as a official function to the integrated flasher.

Checks:
- attic ram (already there)
- slot 1 (and slot 2?) contain a MEGA65 core
- core file needs to be a MEGA65 core!

Questions to be answered:
`I am aware that I void my warranty if I flash slot 0 without a JTAG interface`
`I hereby confirm that I am able to boot the system by selecting slot 1 from the flash menu`
The user needs to type `yes` to both of them and confirm each entry with return. We (deft, martin, I) are positive that this will be enough to not be accidentally triggered.
In addition the Slot 0 Flash functions stays on the (currently secret) MEGA-something combination, so that it is not triggered by accident.

So in total three prerequisites need to be fulfilled and the user needs to confirm two question by typing YES

Recovery:
if slot 0 is broken because of some mishap, the system will boot slot 1 in a short while and you can use NO-SCROLL to enter the Slot 1 flasher.

dansanderson, 2022-10-04:
We should avoid implying the existence of a warranty if there isn't one. I think we just want to emphasize that this is a technically complicated procedure that may fail, and failure will require a recovery process.

> You are about to replace the core in slot 0. Do not turn off your machine or interrupt this process.
> 
> If the process is interrupted or the replacement fails in some way, your MEGA65 will not boot normally: it will blink blue lights for 30 seconds, then boot to core 1. You can restart the core 0 flashing process by holding the NO SCROLL key during these 30 seconds.
> 
> It appears you have a MEGA65 core installed in slot 1. You should test that the core in slot 1 boots correctly before proceeding.
> 
> Type "I UNDERSTAND" to confirm that you understand this message, or press Return to abort: [I UNDERSTAND]
> Type "CONFIRM" to confirm that you are able to boot the core in slot 1, or press Return to abort: [CONFIRM]
```

lydon42, 2022-10-04:
Three words: to many characters!

dansanderson, 2022-10-04:
[image]

```
You are about to replace core 0.

If this process fails or is interrupted,
core 0 will fail to boot and MEGA65 will
blink blue lights.

If this happens, hold the NO SCROLL key
for 30 seconds to restart this process.
This requires a working MEGA65 core in
slot 1.

Be sure to confirm that core 1 works
before proceeding!

  * I have read and understand this
    message.

  * I have tried booting with core 1
    and it works.

  Type CONFIRM to proceed, or press
  Return to abort: CONFIRM_
```

dansanderson, 2022-10-04:
If you need fewer bytes, let me know how many bytes you're willing to spare. :)

lydon42, 2023-03-28:
Extra ideas:
- make it that there is NO file selection, but a fixed `FIRMWARE.COR` is loaded
- add `FIRMWARE.CRC` which contains a CRC checksum
- add checksum calculation which calculates the checksum from attic ram contents and compares to `FIRMWARE.CRC`
- add erase first sector on any failure after flash has started
- add reprogram fpga from slot 1 after flash failure

lydon42, 2023-09-24:
This is in 683-cartflash but still needs finishing touches.

closed 2024-02-24 by lydon42