MEGA65/mega65-core #631: SID 8580 won't play certain notes on first invocation

bubbob42, 2022-10-22:
**Test Environment (required)**
 - Platform: DevKit, MEGA65
 - Core Commit:  93d55f08
 - ROM Release:  920377

**Describe the bug**
When SID8580 is enabled, a noise wave won't be played on first attempt.

**To Reproduce**
1. Set SID to 8580 (6581 is not affected)
2. Switch on the machine (cold boot required)
3. enter: SOUND3,5000,10,2,0,0,3

**Expected behavior**
A noise signal should be played on channel 3. It will only play when repeating the command. This behaviour is present for all SID channels.

johnwayner, 2022-11-05:
This issue only effects the NOISE waveform.  The issue is resolved once a different waveform is selected and gated on (or Basic's SOUND interrupt shut's down a voice by writing $08 (enabling TEST) to its control register).

@gardners The LFSR is stalled out initially due to it relying on wave_out (which will be all zeroes) for part of its value.  Most importantly wave_out[4] will be zero to start and that needs to be 1 to prevent the stall.  Basic sets the TEST bit of the CONTROL register which unstalls the LFSR:

`(lfsr_noise[17] ^ lfsr_noise[22] ^ reset ^ 'test_ctrl)`

Enabling another waveform unstalls the LFSR as well by feeding non-zero values into wave_out.

There is a [more traditional implementation](https://github.com/MEGA65/mega65-core/blob/93d55f0841d48f52cec420621f692bf6ece36fed/src/verilog/sid_voice_8580.v#L155) of an LFSR commented out below [the current one that uses wave_out](https://github.com/MEGA65/mega65-core/blob/93d55f0841d48f52cec420621f692bf6ece36fed/src/verilog/sid_voice_8580.v#L135) that seems to work.

I assume there was a reason for the more complicated one.  Perhaps it simulates the actual 8580?  I have no idea.

Anyway, it can be fixed by reverting back to the commented code or otherwise ensuring the LFSR doesn't ever equal all zeroes.

I can provide a pull request to revert to the commented code, but I don't want to do that without understand why it is the way it is.

lydon42, 2024-09-01:
@bubbob42 Is this fixed?

closed 2024-09-07 by lydon42