MEGA65/mega65-core #634: HYPPO should restore ROM area, even if marked write-protected

gardners, 2022-11-01:
**Test Environment (required)**
You can use [MEGA65INFO](https://files.mega65.org?id=3eeb7d82-8bd7-44b7-82b6-cef29d123b0e) to retrieve this.
 - Platform: ALL
 - Core Commit: ALL
 - ROM Release: N/A

**Describe the bug**
Modifying ROM memory from the freezer has no effect if ROM write-protect is enabled. The root cause is the unfreeze routines don't un-protect the ROM before unfreezing.

**To Reproduce**
1. Freeze
2. Modify some string in the ROM, e.g., the C64 mode basic banner around $2E400
3. Exit freezer
4. GO 64 or SYS64738 to restart C64 mode (don't use reset button, as that will reload ROM)
5. Observe that your changes did not survive unfreezing

**Expected behavior**
The changes made to ROM banks should be effective after unfreezing

**Screenshots**
N/A

**Additional context**
This is why loading ROMs in the freezer never worked.

gardners, 2022-11-06:
believed fixed, but requires confirmation.

lydon42, 2023-03-11:
ROMLOAD does work with this fix.

lydon42, 2023-12-28:
This seems to be broken again in latest development.

lydon42, 2023-12-28:
Question for @gardners: is it really possible to write a m65 tool unit test for this, so this can be regression tested? The unit test would need to get the system into hyppo mode and then try to write ROM space... (because of the fix-test-required label).

commit 0c1e9069 (2022-11-01): fix #634
commit 3953c38b (2022-11-01): fix #634 by making all RAM writable by hypervisor
closed 2024-02-24 by lydon42