MEGA65/mega65-core #644: SD card utility should erase config sector; offer complete formatting options

dansanderson, 2023-01-09:
A recent Discord discussion and troubleshooting session with kibo, Maurice, and KiDra in #peripherals suggests that the built-in SD card utility does not erase important portions of the card during formatting. It may be possible to have a card in a broken state, in this case a corrupt config sector, and not be able to completely reset it with the SD card utility.

A workaround is to completely erase (write all zeroes to) the SD card with a PC before using the SD card utility to set up MEGA65 stuff. (The Windows tool https://rufus.ie/en/ was mentioned.) We should improve the built-in SD card utility so that this is not required. At least, we should reset the important sectors that need to be reset for the MEGA65's purposes as part of formatting.

If the config sector is intentionally left alone during formatting, we should make resetting it an option of the SD card utility. It appears to be insufficient to assume the user will go through on-boarding and rewrite the config sector: under some circumstances, formatting the card does not re-trigger onboarding.

dansanderson, 2023-01-21:
Moved to mega65-fdisk https://github.com/MEGA65/mega65-fdisk/issues/19

closed 2023-01-21 by dansanderson

gardners, 2023-01-21:
Can't the configure utility be used to reset the config sector? If not, that's probably the right place for this