MEGA65/mega65-core #651: AUDIO_DMA: audio samples sound very distorted

lydon42, 2023-01-20:
Issue #228 introduced an `audio_dma_saturation_enable` switch, which seems to be enabled by default.

This apparently results in distorted audio.

Todo:
- [ ] change default to off (?)
- [ ] document feature so it shows up in iomap.txt and user guide

M3wP, 2023-01-20:
I am working in the VHDL to determine why it is an issue since the feature is actually desirable but shouldn't be causing so much trouble.  The problems really shouldn't be happening.

I'd like to see the default setting be off, though.

M3wP, 2023-02-01:
I'd like to make a note that the version merged did not include fixes after discussion with kibo.

lydon42, 2023-05-07:
So is here still something missing @M3wP and @ki-bo ?

ki-bo, 2023-05-07:
I will need to investigate and look up past conversations, as I don't remember 100% anymore.

M3wP, 2023-05-07:
Sorry, yes.  There were a few changes that @ki-bo made with me in discussion that weren't in the changes that were applied.

I've attached what I have.  @lydon42, you should be able to rescue them from the attached file.

[gs4510.zip](https://github.com/MEGA65/mega65-core/files/11416021/gs4510.zip)

lydon42, 2024-01-23:
Found this while doing the Changelog.md.

Comparing this gs4510.vhdl to release-0.96 is hard, because there are a lot of changes. It seems the only thing that is really different are the parts which start with ` -- overflow: so saturate instead`.

I don't know if this is still an issue and needs to be fixed or if this is already fixed. Can someone check this too?

M3wP, 2024-01-24:
It should be largely fixed however the implementation that was taken is not the last one I had made.  I made additional changes based on some conversation with kibo but they were not copied to the code base afaik.

I still have the complete changes.  It is some time ago that this was done but I should be able to assist in reconstructing the code.

commit 08bf8228 (2023-01-21): pull in m3wp's audio DMA fixes #651