MEGA65/mega65-core #652: SD card utility should create a README.txt file along with system files from the core

dansanderson, 2023-01-21:
**Is your feature request related to a problem? Please describe.**
Users need more opportunities to learn about SD card management. For example:

1. The purposes of the files installed by the SD card utility
2. Where to download bundled software (GEOS, etc.) and updates
3. Why and how to format the card with the MEGA65 SD card utility
4. How to work around issues with incomplete FAT32 support (fragmentation), using mega65_ftp/etherload
5. Tips for dealing with spurious files installed by the host OS (especially macOS) or long filenames

**Describe the solution you'd like**
Create a plaintext document named `README.txt` containing this information. Embed it in the core as one of the files it populates via the SD card utility. Include it in the bundled SD card image.

**Describe alternatives you've considered**
Alternatively, the README.txt file could be a short file with a reference to a chapter in the manual, or a mega65.org short URL that redirects to a Filehost or Wiki article. The goal is to present the user with this information at the point it is useful, i.e. when they mount the SD card on their PC. A user will probably follow a link found in this file.

dansanderson, 2023-01-21:
Moved to https://github.com/MEGA65/mega65-fdisk/issues/18

lydon42, 2025-04-28:
As this needs to be a file added to the core, it is a core issue, closing fdisk issue instead.

lydon42, 2025-04-28:
Adding is easy. please write something and place it into release-build as `README-sdcard.txt` (or .md, as you prefer).

lydon42, 2025-07-21:
379e45f
f48d68a

closed 2025-07-27 by lydon42