MEGA65/mega65-core #656: IRQ feature in AudioDMA

M3wP, 2023-02-01:
**Is your feature request related to a problem? Please describe.**
Use of AudioDMA is highly desirable but uses quite significant amounts of memory.  Buffering can be used to reduce this footprint but is difficult to achieve reliably without IRQ notifications.

**Describe the solution you'd like**
A "low water" alarm should be available for each channel and flags to enable and represent the condition of having entered into a specific region of the sample data.

**Describe alternatives you've considered**
Polling the playback position is an alternative but you want to poll for the loop point (end point) and this is particularly difficult to do without doing only this and in large buffers.

**Additional context**
I have implemented the feature and will make it available after more thorough testing.

lgblgblgb, 2023-02-01:
@M3wP If possible, please make a test program available using this feature, would be great for Xemu too, since I need to add this feature there as well, and it's nice then if there is some test material available. Thanks in advance!

M3wP, 2023-02-01:
[probe.zip](https://github.com/MEGA65/mega65-core/files/10561091/probe.zip)
Here you will find my test program for the IRQ feature.  It shows setting and reading the IRQ enable and event flags and how to use the trigger value to trigger the IRQ at the half-way point.

It has become a little "hacked" since I have been using it to track down problems in Manche v2 which also will make use of this feature but should be useful still as a reference.

lydon42, 2024-08-18:
This is #811

closed 2024-08-18 by lydon42