MEGA65/mega65-core #659: Collision detection does not work in "Graphic Mode" for basic programs

thecodecrusher, 2023-02-02:
You can use [MEGA65INFO](https://files.mega65.org?id=3eeb7d82-8bd7-44b7-82b6-cef29d123b0e) to retrieve this.
 - Platform: MEGA65
 - Core Commit: 93d55f0
 - ROM Release: V920377

M65 does not detect if a sprite on the Background e.g. a line if you are in then graphic mode. 
Issue appeared in a Basic program. But only on Mega Hardware. Xemu works fine.

**To Reproduce**
1. open a screen
2. draw a line
3. activate "COLLISION" interrupt for sprite-data  
4. move a sprite over the line (step2)
5. and see that the collision routine is not entered

lydon42, 2024-08-18:
@dansanderson is this a known issue? Is it VICIII or ROM (I suspect VIC)?

dansanderson, 2024-08-18:
It's a VIC issue.

It would make sense for sprites to collide with bitplane graphics, and we should probably implement it. IIRC the VIC-II will collide sprites and high-res graphics. I would be curious to know if an actual VIC-III does this correctly with bitplane graphics, or if it is at least specified to do so.

Just to confirm the test program above, you can comment out lines 100 and 110 and add a line: `115 PRINT "{clr}{6 down}{30 Mega+@}"` This is the text mode equivalent of drawing a line, and the sprite will trigger the collision trap as it runs across it.

lgblgblgb, 2024-08-18:
@dansanderson As far as I remember (!!!) I've read somewhere that sprite-background collision on C65 only "senses" bitplane number 2 if it's a bitplane mode. I can be wrong, this is my memory (or was it with the sprite priority?). But really, it's only a faint memory and can be something else.

Ah, found it, in `c65manual.txt`:

_Note also that, in bitplane, mode, sprites will only collide with "background"  data  which  has  bits  "on"  in  bitplane 2.  All other bitplanes will NOT cause a sprite-to-background-data collision._

dansanderson, 2024-08-18:
oh wow, good info! Maybe we need to reproduce this issue more specifically then, in case someone already thought to implement this for the VIC-IV and we're just not testing it correctly.

lydon42, 2024-08-18:
Wow, yes, this is really interesting. So we need a test which has data on bitplane 2 and some data on a different bitplane. And perhaps a border flicker to show... if this works we are conformant. And then I would close the issue, as we probably don't want to put more work into bitplane graphics than needed to get the VIC-III feeling. For everything else we have VIC-IV and RRB.

lydon42, 2024-08-18:
```
 100 SCREEN 320,200,3                                                             
 110 PEN 3:LINE 0,100,319,100                                                     
 120 PEN 5:LINE 0,50,319,50                                                       
 130 PEN 7:LINE 0,150,319,150                                                     
 140 COLLISION 2, 300                                                             
 150 SPRITE 1,1:MOVSPR 1,120,0:MOVSPR 1,0#2                                       
 160 GETKEYA$                                                                     
 170 COLLISION 1                                                                  
 180 SPRITE 1,0                                                                   
 190 SCREEN CLOSE                                                                 
 200 END                                                                          
 300 BORDER 3:PRINT BUMP(2):BORDER 6                                              
 310 RETURN                                                                       
```
This test program does also not collide. So either I don't know what bitplane 2 is, or there is no collision detection. A quick scan of the vic vhdls also does not show anything that looks like it.
[bumpbp.zip](https://github.com/user-attachments/files/16651442/bumpbp.zip)

Rhialto, 2024-08-19:
The bitplane 2 thing seems to be meant as a feature. Elsewhere in the document it says:
>  Bitplane 2 is the foreground/background plane used for sprite/background collision detection and priority.

> The attributes are only applied to bitplane 2, which is also the  foreground/background  plane  for  sprite collisions and priority purposes.

(How this interacts with the also mentioned FGBG bit in the colour palette... dunno. It would make sense to also have a sprite-foreground collision and use FGBG for that but that would be a new thing which isn't mentioned.)