MEGA65/mega65-core #663: Add Option for 4:3 Aspect ratio HDMI Output

michaelniotis, 2023-02-05:
**Is your feature request related to a problem? Please describe.**
No problem

**Describe the solution you'd like**
Being able to select original 4:3 Ratio for HDMI Output

**Describe alternatives you've considered**
Currently changing Aspect Ratio from VGA Monitor via monitor options

**Additional context**
Use either a 4:3 toggle switch or add additional resolution options with the 4:3 Ratio in the description

lydon42, 2023-02-06:
The output of the MEGA65 is 4:3 (or 3:4), as it is 480 or 576 (don't remember if it is i or p).

Do you perhaps mean to add a mode that is proper 720p/1080p and has the 4:3 screen inside of this wider frame?

michaelniotis, 2023-02-06:
Something similar that the C64 Core for the M65 does so that you do not have to adjust the aspect ratio on your screen [](https://github.com/MJoergen/C64MEGA65)

Here is what they claim they do: 

_HDMI: The core outputs 1280×720 pixels (720p) at 50 Hz and HDMI audio at a sampling rate of 48 kHz by default. This is supported by a vast majority of monitors and TVs. The 4:3 aspect ratio of the C64's output is preserved during upscaling, so that even though 720p is a 16:9 picture, the C64 looks pixel perfect and authentic on HDMI._