MEGA65/mega65-core #667: Allow Hyppo and Freezer to swap writeable charset with default temporarily

dansanderson, 2023-02-08:
**Problem**
It is currently possible to write to VIC-IV Character ROM at FF7E000-FF7EFFF. This changes the default character set, which is used by the Hypervisor, the Freezer, and the main system if CHARPTR is set to the special value $0001000. The special value is not documented, but it is the default on boot. Character ROM *is* documented as "write-only," and BASIC and some programs customize the character set by writing to this memory instead of setting CHARPTR to RAM.

A side effect of writing to character ROM is that Hyppo and Freezer displays use whatever modifications were set by a program, possibly becoming unreadable. Hyppo also uses the customized charset for the boot display when the user presses the reset button.

**Proposed solution**
Hyppo should always use the default PETSCII character set *and* preserve any modifications that were written to VIC-IV char ROM by a program in cases when it's possible to resume the program. For example, after a program writes to VIC-IV char ROM, entering the Freezer should switch to the default PETSCII charset, and resuming from the Freezer should restore the modified charset.

One possible implementation: add new FFxxxxxx actually-read-only ROM for storing the PETSCII charset, then have Hyppo/Freezer use CHARPTR to select it on entry and restore CHARPTR on exit. Formalize CHARPTR=$10000 as "use the writeable VIC-IV char memory." This would give BASIC and programs the ability to force PETSCII without overwriting VIC-IV char ROM, as well as a way to access ROM source of truth for the original PETSCII charset even after lower ROM has been clobbered.

**Alternatives considered**
We could renege on making VIC-IV char ROM writeable. It would always contain the PETSCII charset. This would break existing programs, and would require rewriting BASIC CHARDEF. If we also make VIC-IV char ROM readable, BASIC and other programs could safely clobber ROM data at 0002D000-0002DFFF for use with CHARPTR, and could restore it any time from VIC-IV char ROM. I suspect the backwards incompatibility would be too conspicuous to allow for this option.

lydon42, 2023-03-12:
In #676 hyppo freeze/unfreeze of the chargen area was implemented.

lydon42, 2023-05-31:
@dansanderson as you can read just above: #676 did add the feature that the FREEZER can read the WOM, so it can be frozen correctly.

dansanderson, 2023-08-07:
This issue requests that Hyppo/Freezer use the default charset when displaying its own UI. It's great that it freezes it now, but to my knowledge we haven't yet implemented Hyppo/Freezer temporarily restoring the default charset for its own use. Presumably the custom changes would come back when it restores, so it just needs one more step to copy PETSCII into char memory just after freezing.

I don't think we answered the question about where Hyppo/Freezer would find the PETSCII charset. There's a ROM location, but programs can technically overwrite that. I recall asking about how char memory is initialized on boot but I don't remember if I got a response. I assume it can just be re-initialized?

Repro steps:
1. `CHARDEF 1,$FF,$FF,$FF,$FF,$FF,$FF,$FF,$FF` : This sets the letter A to a solid block.
2. Hold Restore then release to open the Freezer.
3. Press F3 to resume.

Expected: The letter A appears normally in the Freezer, then returns to a block shape after resuming.

Actual: The letter A appears as a block shape in the Freezer. (True as of 684-cartflash,20230612.07,cae3b0b.)

lydon42, 2023-08-07:
Pressing F14 loads the default charset "CHARSET.M65" or the one inside "MEGA65.ROM".

lydon42, 2023-08-07:
Not a core issue.

closed 2023-08-07 by lydon42