MEGA65/mega65-core #673: MEGA+SHIFT+POUND keypress should only be allowed via physical keyboard input

ki-bo, 2023-03-09:
**Is your feature request related to a problem? Please describe.**
The remote network access (which can be a security risk when turned on unintentionally and/or in insecure environments) needs to be activated until the next cold start via MEGA+SHIFT+POUND. As this is a security related feature, that keypress should only be possible to trigger via physical keyboard.

**Describe the solution you'd like**
Disable MEGA+SHIFT+POUND keypresses via synthetic key events.

gardners, 2023-03-12:
See c200b92ccb0b0eaf951e218c69495971642aa305

It turns out I made it just SHIFT+POUND with this commit.

gardners, 2023-03-12:
Confirmed working with SHIFT+POUND on this commit.

lydon42, 2023-12-23:
We keep SHIFT-POUND for batch 3 and discuss possible more complex security changes afterwards.

lydon42, 2024-01-03:
This stays as it is for the release.

closed 2024-01-05 by ki-bo