MEGA65/mega65-core #674: A 1581 does not work anymore once the freezer was accessed

Schwefelholz, 2023-03-10:
**Test Environment (required)**
You can use [MEGA65INFO](https://files.mega65.org?id=3eeb7d82-8bd7-44b7-82b6-cef29d123b0e) to retrieve this.
 - Platform: MEGA65
 - Core Commit: 93d55f0
 - ROM Release: 920377

**Describe the bug**
A 1581 connected to the serial IEC bus does not work anymore once the freezer was accessed. Only a power cycle of the MEGA65 will resolve this issue. Neither a computer reset nor power-cycling the 1581 has any effect.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a 1581 (set to U11 to be able to access it at all)
2. Insert a formatted disk
3. `DIR U11` correctly displays the directory
4. Enter the freezer
5. Leave the freezer (no action necessary)
6. `DIR U11`: The activity indicator and the drive motor of the 1581 come on, but the directory is not displayed; the computer hangs
7. Run/Stop - Restore ends the hanging process, but no access to the 1581 will work until the computer is turned off/on.

**Expected behavior**
A 1581 should not be affected by entering/leaving the freezer, as is the case for a 1541 or 1571.

**Additional context**
The effect on `DIR` is just an example. `DLOAD` or `DSAVE` will hang as well. Reading the error channel, however, seems to be possible.
The effect is also there if the 1581 was powered on _after_ the freezer was entered/left.
A 1571 is proven to be unaffected.

eilers, 2024-09-01:
I followed the test case and got the following results:

R3A, Core:  0.96 6d764c4 (with hardware IEC), and ROM 920407 (with IEeC support VHWIEC4): ✅
R3A, Core:  0.96 3c10488 (without hardware IEC), and ROM 920408 (latest beta release): ❌
R3A, Core:  0.96 6d764c4 (with hardware IEC), and ROM 920408 (latest beta release): ✅ 

R6, Core:  0.96 6d764c4 (with hardware IEC), and ROM 920407 (with IEeC support VHWIEC4): ✅
R6, Core:  0.96 3c10488 (without hardware IEC), and ROM 920408 (latest beta release): ❌
R6, Core:  0.96 6d764c4 (with hardware IEC), and ROM 920408 (latest beta release): ✅ 
R6, Core: Development #124 3421b63, and ROM 920408 (latest beta release): ❌
Even the latest development core #124 still has this problem. Thus, the Core with HW IEC support fixes it.

lydon42, 2024-09-01:
@Schwefelholz you can get a IEC HW Accel core for testing on [builder](https://builder.mega65.org/job/mega65-core/job/736-hardware-iec-accel/).
It seems the work in done in #736 does solves this problem.

Schwefelholz, 2024-09-01:
I confirm it's working for me as well using [64764c4](https://github.com/MEGA65/mega65-core/commit/6d764c4f97c13db2ce3eb88a6c6df70e5d9ac149) and ROM 920395 or later.
Thank you!

lydon42, 2024-09-01:
Leaving this open & staged, as 736 is not merged yet.