MEGA65/mega65-core #675: Text screen does not display the very leftmost pixel column

Schwefelholz, 2023-03-10:
**Test Environment (required)**
You can use [MEGA65INFO](https://files.mega65.org?id=3eeb7d82-8bd7-44b7-82b6-cef29d123b0e) to retrieve this.
 - Platform: MEGA65
 - Core Commit: 93d55f0
 - ROM Release: 920377 (and also original Commodore ROM 911001)

**Describe the bug**
The leftmost text column seems not to display its leftmost pixel column. E.g. Typing a "K" as a first letter of a line and placing the cursor on it looks as if the Ks vertical bar is on the left edge of the screen position. If you type a "K" on the second position and put the cursor on it there seems to be an empty pixel column on the left edge.
This is even better to see if the border color is set differently from the background color.
I checked this with Commodore ROM 911001 as well and it shows the same effect.

**To Reproduce**
Steps to reproduce the behavior:
1. Type a reverse "K" as the first letter on a line. It looks as if the Ks vertical bar is on the left edge of the screen position.
2. Type a reverse "K" on the third position. There seems to be an empty pixel column on the left edge.

**Expected behavior**
The K should be displayed equally on first and other screen positions.

**Screenshots**
[image]

**Additional context**
Interestingly, taking a screenshot with M65Connect shows the character as is to be expected:
[image]

Schwefelholz, 2023-03-10:
First findings by @ki-bo logged on Discord:

> First conclusions from my side:
> - The ROM is doing all as it should
> - $D016 XSCL is not being applied directly, instead it is triggering a recalculation of the VIC-IV registers for border position/size and text/char generator position and size (supporting fact: XSCL seems to do nothing if you disable hot registers)
> - The mentioned C65 VIC issue with one-pixel-off in H640 mode is modelled in the translation from $D016 to VIC-IV registers only (as it seems)
> - The VIC-IV has some bugs in applying TEXTXPOS and SDBRDWD registers
> - Typically, if the latter two are set to the same value, everything should be correct. But instead, the char generator is one pixel off from the border. Which one of them is wrong is not clear to me, yet.
> - The last pixel column of the char generator seems to be not drawn anymore (it seems it is only drawing 639 pixels instead of 640). This is often not visible because at the same time the border is drawn one pixel too narrow on the right side (in H640 mode)
> - Will need to dig deeper into viciv.vhdl to understand all these, but don't have more time today
> -I guess H640 mode is not yet used a lot, and many still use 320 pixels for their work, maybe that's the reason I haven't seen complaints on this, yet

ki-bo, 2023-03-14:
This is only happening when crt emulation is enabled

closed 2023-05-25 by ki-bo