MEGA65/mega65-core #676: Get rid of charset WOM area and make it a ROM mirror

lydon42, 2023-03-11:
Currently the freezer has no way of reading the current charset, and as such the charset can not be frozen.

@gardners suggested that I ask him to replace the WOM charset region by a mirror of the ROM charset instead...

I am oblivious to the consequences of that :)

dansanderson, 2023-03-11:
Please also consider this alternate proposal I filed a month ago: https://github.com/MEGA65/mega65-core/issues/667

lydon42, 2023-03-12:
Test looks good, works well!

closed 2023-05-31 by lydon42

dansanderson, 2023-05-31:
Can you summarize the accepted change? The issue title implies that writing to what used to be the char WOM area now overwrites the CHARC region of chip RAM (with write-only protection ignored). If that's true, that would have some significant backwards incompatible consequences.

If we went a different way, I'd like to understand it so I can update the Memory chapter.

lgblgblgb, 2023-05-31:
@dansanderson @gardners I'm interested too ... As far as I know, the situation: the "char ROM" inside the "usual" ROM (which would have been used by C65) does (did?) not do anything at all. The real thing was the "WOM" (Write-Only-Memory). As far as I guessed, the change only is making the WOM readable (but still no connection between the "WOM" and the 128K's ROM content, and no change there), thus shouldn't be called WOM anymore :) So the "main ROM" is not affected at all. But this is only a guess of mine.

lydon42, 2023-05-31:
The change is that the FREEZER can read the WOM, making freezing it possible.

commit f169110c (2023-03-12): more plumbing fixes #676