MEGA65/mega65-core #677: HYPPO: add better disk image directory handling

lydon42, 2023-03-11:
Currently only the filename of a disk image attached is stored in the process structure. If an image was loaded from a subdirectory, it is never possible to restore it after a freeze.

In addition hyppo could also store some UID of the sd card, so the entries can be invalidated if the wrong disk is inserted. Or better not?

Related FREEZER issues:
 - https://github.com/MEGA65/mega65-freezemenu/issues/62
 - https://github.com/MEGA65/mega65-freezemenu/issues/63

lydon42, 2023-03-11:
Perhaps it is also possible to just store disk UID plus some pointer onto the disk? Not sure if this is feasible with FAT. Problem with path storage: those can get quite long...

lgblgblgb, 2023-03-12:
In theory at FAT level, directory can be identified with only a single number: the starting cluster of the given directory (if it's the root directory, the starting cluster simply the cluster number of the start of the root directory but there is nothing special in that on FAT32 - though it is on FAT12/16, but it's not scope here fortunately). There is not even needed to store the textual representation of the full path, as every directory should have '..' entries to be followed to query the path then if needed. The disadvantage of this, if someone wants to see the path, not only make it working. Then the mentioned '..' following should be done to get back the path (in reversed order) till hitting the root directory. I am not sure though, how HYPPO handles these things ...

lydon42, 2023-09-03:
closing in favor of #729

closed 2023-09-03 by lydon42