MEGA65/mega65-core #683: MEGAFLASH: add ability to select a slot that is booted with a cartridge

lydon42, 2023-03-15:
With the recent C64 alpha core that supports cartridges, it would be nice if the MEGA65 can boot a specific slot if a cartridge was detected.

Plan:
- [ ] add flag so one slot can flagged as the cartridge slot
- [ ] detect cartridge on startup and select cartridge core slot instead of default slot to boot
- [ ] add ability to change cartridge core slot in MEGAFLASH

Thoughts:
- MEGAFLASH will only show 31 characters of the NAME and VERSION fields, so we can easily use the 32nd character to store the cartridge slot selection.
- reflashing of the first block of a slot can be used to change this flag, this should be fairly quick
- for testing bit2core can implement the flag directly into a COR
- if multiple slots have the flag, the lowest slot wins!

lydon42, 2023-03-15:
kernel64.src:4114+ does check exactly for CBM80 at $8004, so checking MEGA65 at $8009 seems to be viable.

lydon42, 2023-03-17:
The `MEGA65` check is not really necessary, we can just do it the C128 way, not lowering /GAME and /EXROM and depending just on a memory signature.
So I will remove this again and rather exchange it with a C128 core detection.

lydon42, 2023-04-10:
[mega65-tools/dev#6ed6071](https://github.com/MEGA65/mega65-tools/commit/6ed6071233d4f73a6a03b62b895c973fe2f8b006) adds bit2core caps/flags features.

lydon42, 2023-04-30:
typo in the commit message: 85f387ef7f7929fba41b849a3788b21f08aff582

lydon42, 2023-05-02:
Current plan:
 - add CRC32 calculation to core integrated MEGAFLASH
 - allow Slot 0 Upgrade from within Core integrated MEGAFLASH

lydon42, 2023-05-04:
Sucessfull borken slot 0 recovery using etherload to push upgrade0.prg!
Everything needed is in latest build of this branch and mega65-tools development (ethload tool).

gardners, 2024-01-28:
Examining problems with R3 board.
Things as I discover them:
1. Hardware-accelerated RDID reading does not work. The old soft-ware based method does.
2. mhx_writef() seems to be going crazy sometimes. I can't, for example, reliably use the flash inspector, because it stops writing to the screen.
2.1. The problem seems to be mhx_saddr gets out-of-bounds somehow.
2.2. mhx_posy doesn't reach 25 by the time cursor is at bottom of screen to scroll, so scrolling doesn't happen, and it will advance past $07E8 and potentially start corrupting things.
3. erase_sector() checks sector size using variable addr, not the address_in_sector argument that is passed to it.  This might be causing the selection of the wrong flash sector size to erase, at least some of the time.
4. read_data() uses hardware-accelerated QSPI reading, which seems to not work on R3.

In short, the nexus of the problem seems to be that the R3 QSPI flash chip and the hardware-accelerated QSPI functions in sdcardio.vhdl don't seem to work together.

gardners, 2024-01-28:
For (2) above I have patches to the flash inspector to reset the mhx cursor to 0,0 which works around the problem.

Need to look more deeply into why the hardware-accelerated QSPI functions are failing on R3.  Likely need to setup a model of the QSPI flash chip under VHDL simulation, and see what is going on.  This will require considerable effort and attention which I may not be able to invest until after the blocking issues for R5/R6 are solved in 0.96, in particular the expansion port issues.

lydon42, 2024-07-02:
Continued in #782

closed 2024-07-02 by lydon42