MEGA65/mega65-core #688: CRT emulation should not be changeable by software

deftmega, 2023-04-13:
Many programs overwrite the flag for CRT emulation, probably accidentally. In my opinion the choice if CRT emulation is on or off should not be made by the coder if an app but only by the user and via config menu. I would ask and propose to invent a lock for that bit after boot time or something similar. Thank you for considering.

lgblgblgb, 2023-04-13:
I would say, it's OK to be switchable, if it's allowed. Ie, there can be a "locking" setting so programs cannot alter it, but there can be use cases when it's indeed useful that a software can turn it on/off, like when testing something etc by a test program without the need to change settings manually for example. Btw, similar wish can be valid idea for other kind settings as well in general ...

tdeval, 2023-04-13:
IMNSHO, I would say as it is to simulate a "hardware" behaviour, I don't see the point of having it software changeable.
I would rather like it being a hypo-only changeable setting, much like the SID emulation, the joysticks swap, etc...