MEGA65/mega65-core #689: Strange blemishes/artifacts when using 640x400 16 colour bitplane graphics mode

gurcei, 2023-04-16:
**Test Environment (required)**
 - Platform: MEGA65
 - Core Commit: release 0.95 (batch2)
 - ROM Release: (i.e. 920377)

**Describe the bug**
Drawing graphics in 640x400 16-colour bitplane graphics mode results in mild blemishes on the screen

**To Reproduce**
Steps to reproduce the behavior:
Either play "Onion Cake and the Hungry Dinosaurs"
- https://files.mega65.org/?id=e15d97b6-af4d-47a0-969a-3a6691034bfd

Or play around with 640x400 16 colour mode directly in BASIC with:
`SCREEN 640, 400, 4`

And then try a few BASIC line, box, circle calls.

lydon42, 2023-05-10:
Example from #701 
[image]

gurcei, 2023-05-12:
I've documented issues with drawing horizontal lines in various colours in this discord thread:
- https://discord.com/channels/719326990221574164/782759365064917065/1106523564326277160

Key insights were:
- Colours 0 to 7 draw fine regardless of which y position the horizontal line is on
- Colours 8 to 15 are problematic and they all share the following behaviour:
  - For y=0 to 5, they draw the horizontal line in the correct colour
  - For y=6, they lose bit3 of their colour index. I.e.:
    - orange becomes black
    - brown becomes white
    - lt red becomes red
    - dk grey becomes cyan
    - grey becomes purple
    - lt green becomes green
    - lt blue becomes blue
    - lt grey becomes yellow
  - For y=7, two orange horizontal lines are drawn on y=6 and y=7

stricq, 2023-07-14:
Yes, I've noticed this as well by modifying the Hopalong program on the demo disk to run in 640x400x16.  Colors 0-7 work fine, but colors 8-15 draw as different colors on approximately every 8th row.

ki-bo, 2023-07-15:
> Yes, I've noticed this as well by modifying the Hopalong program on the demo disk to run in 640x400x16. Colors 0-7 work fine, but colors 8-15 draw as different colors on approximately every 8th row.
> 

Can you provide the modified sample program for reproduction?

dansanderson, 2023-07-16:
This is interesting:

```
5 screen 640,400,4
6 pen 9
10 for y=0 to 64
20 line 0,y,300,y
30 getkey a$
40 next y
50 screen close
```

Expected behavior: It draws a single brown horizontal line at the top of the screen. Each time you press a key, it draws another brown horizontal line one pixel below it, gradually drawing a brown rectangle.

Actual behavior: The first thing you see is a white line in row 0 and an orange line in row 1. That's just from the first `LINE` command, y=0. On the next keypress, y=1, both lines turn brown. y=2: now there's a white line on row 2 and an orange line on row 3. y=3: both lines turn brown. It's like this all the way down. (I don't see different behaviors every eight lines in this example.)

[image]
[image]

Reproduced here with core 683-cartflash,20230612.07,cae3b0b, ROM 920383.

dansanderson, 2023-07-16:
... Trying the other colors C, on the even lines I'm seeing C-8 on the intended line and orange (8) on the line below.

If I STEP 2 from Y=0 (draw even rows), I get C-8 on Y and 8 on Y+1. If I STEP 2 from Y=1 (draw odd rows), they're flipped, including the first line drawing 8 on line 0 and C-8 on line 1.

ki-bo, 2023-07-16:
Seems like this was already the case with batch 2 release core.

stricq, 2023-07-16:
`100 REM HOPALONG FRACTAL DEMO FOR MEGA65
110 REM STEPHAN KLEINERT, AUG 2020
120 :
125 MX=320:MY=200
130 POKE 0,65
150 PRINT"** HOPALONG FRACTAL FOR THE C65/M65 **"
160 PRINT"   S. KLEINERT, AUGUST 2020"
170 PRINT"WHILE RENDERING, PRESS ANY KEY TO ABORT"
180 PRINT"CURRENT FRACTAL AND START A NEW ONE, OR"
190 PRINT"PRESS Q TO QUIT."
200 PRINT:PRINT"PRESS ANY KEY TO START!"
210 GETKEY A$
220 GRAPHIC CLR
230 SCREEN DEF 1,1,1,4
240 SCREEN OPEN 1
250 SCREEN SET 1,1
255 BORDER 0
260 GOSUB 900
350 DO
360 N=0 : REM NUMBER OF POINTS
370 P=INT(RND(.)*15)+1 : REM CURRENT PEN
380 SCREEN CLR 0
390 PEN 0,P
400 A=-200+(INT(400*RND(3)))
410 B=-50+(INT(100*RND(3)))
420 C=-200+(INT(400*RND(3)))
430 X=0:Y=0
440 DO
450 N=N+1:IF N=3072 THENN=0:P=P+1:PEN 0,P:IF P=15 THEN P=0
460 NX = Y+SGN(X)*SQR(ABS(B*X-C))
470 NY=A-X
480 X=NX
490 Y=NY
492 IF X+MX < 0 OR Y+MY < 0 THEN 510
494 IF X+MX > 640 OR Y+MY > 400 THEN 510
500 DOT ABS(X+MX),ABS(Y+MY)
510 GET A$
520 LOOP WHILE A$=""
530 LOOP WHILE A$<>"Q"
540 SCREEN CLOSE 1
550 PALETTE RESTORE
560 PRINT"THANK YOU & HAVE A NICE DAY."
570 END
600 REM ORIGINAL 8 COLORS
610 PALETTE 1,0,0,0,0
620 PALETTE 1,1,0,12,3
630 PALETTE 1,2,15,0,9
640 PALETTE 1,3,0,6,15
650 PALETTE 1,4,8,0,9
660 PALETTE 1,5,12,15,1
670 PALETTE 1,6,15,0,15
680 PALETTE 1,7,4,3,12
690 PALETTE 1,8,12,11,5
700 RETURN
800 PALETTE 1, 0, 0, 0, 0
805 PALETTE 1, 1,15,11, 9
810 PALETTE 1, 2,15,12, 9
815 PALETTE 1, 3,15,14, 9
820 PALETTE 1, 4,15,15, 9
825 PALETTE 1, 5,12,15, 9
830 PALETTE 1, 6, 9,15, 9
835 PALETTE 1, 7, 9,15,12
840 PALETTE 1, 8, 9,15,15
845 PALETTE 1, 9, 9,12,15
850 PALETTE 1,10, 9, 9,15
855 PALETTE 1,11,10, 9,15
860 PALETTE 1,12,14, 9,15
865 PALETTE 1,13,15, 9,15
870 PALETTE 1,14,15, 9,12
875 PALETTE 1,15,15, 9,10
880 RETURN
900 PALETTE 1, 0, 0, 0, 0
905 PALETTE 1, 1,15, 0, 0
910 PALETTE 1, 2,15, 4, 0
915 PALETTE 1, 3,15, 9, 0
920 PALETTE 1, 4,15,13, 0
925 PALETTE 1, 5,15,15, 0
930 PALETTE 1, 6,10,15, 3
935 PALETTE 1, 7, 0, 8, 0
940 PALETTE 1, 8, 0, 8, 8
945 PALETTE 1, 9, 4,14,13
950 PALETTE 1,10, 8,12,14
955 PALETTE 1,11, 0, 0,15
960 PALETTE 1,12, 3, 0, 8
965 PALETTE 1,13,13, 7,13
970 PALETTE 1,14,15, 0,15
975 PALETTE 1,15,15,11,11
980 RETURN
`

dansanderson, 2023-07-16:
Correct, this is an old issue, not a recent regression.

gurcei, 2024-12-08:
I shared some thoughts on discord that possible cause might be a timing issue, getting through such a long pipeline of activity (8 sprites, 8 bitplanes):
- https://discord.com/channels/719326990221574164/977076334297743360/1315389556199260311

I've heard the vhdl gurus speak of using timing constraints to tighten up such things, but I'm not skilled enough in vhdl to know what to do.

Maybe other vhdl gurus in the community might have some idea on what could be done.

gurcei, 2024-12-10:
Ok, I tried removing the sprites from the pipeline to see if that helped resolve the issue, no it didn't, so it's not that.

In other news, I finally understand why I couldn't see the blemish pattern dddaaannn described in an earlier comment. He was in NTSC mode, and I was in PAL mode. Seems like the blemish pattern was different depending on that choice. I left a vid on discord of this:
- https://discord.com/channels/719326990221574164/977076334297743360/1316126421961412660

gurcei, 2024-12-21:
Making some good progress on this issue, some more details in this discord thread:
- https://discord.com/channels/719326990221574164/977076334297743360/1320031889775398912

The workaround/solution is still in a fairly early experimental phase and I'll try clean it up and test it out more thoroughly in the coming days.

gurcei, 2025-01-08:
Good news, the solution has been merged into development, will close this one off now :)

https://github.com/MEGA65/mega65-core/pull/850

closed 2025-01-08 by gurcei