MEGA65/mega65-core #692: Easyflash menu works but title screen does not show and titles crash after being chosen

deftmega, 2023-04-18:
With the newly fixed ultimax cartridge functionality the easyflash carts start again in the MEGA65 core. However, with dcm.jar generated cart menus a) the title screen will be skipped and b) it is not possible to start any item from the menu (instant crash with blue native screen colors).

This fully worked with the pre-series machines until late summer 2019. I have just tested an r2 core dated 20190701 and the title screen appears and I can run any item from the menu just fine (if its compatible to the c64 side of the core).

I strongly believe that Paul broke said things on November 1st 2019 with changes to the expansion port controller ([expansion_port_controller.vhdl](https://github.com/MEGA65/mega65-core/commits/8623487cc7548a11a46763217828f4e4cf6fad98/src/vhdl/expansion_port_controller.vhdl)) 

I have attached such a game compilation as easyflash flashable disk images.

[deftcart1.zip](https://github.com/MEGA65/mega65-core/files/11263625/deftcart1.zip)
[image]

deftmega, 2023-04-18:
it might have happened [here](https://github.com/MEGA65/mega65-core/commit/8dcbc32975e5195b45709bbe27f4a6634465304f)

deftmega, 2023-04-24:
I got some additional information: since the c64forMEGA65 core recently had the same problem, I asked sy2002 to tell me what it was when he found it. Here is his info in German:

Michael hat den Deftcart Fehler gefunden (also warum es in HARDWARE ging und in der Simulation durch CRT Files nicht): Wir haben in der Software Simulation bisher nicht supported, dass manche Cartridges RAM an der Stelle $DExx und $DFxx unterstützen (jeweils 256 byte). Das EasyFlash nutzt das RAM dort und Diskloader Routinen unterzubringen. Man muss also dieses RAM lesen und schreiben können. Vielleicht hast Du schon gemerkt, dass beim Deftcart im Hintergrund automatisch "LOAD <SPIELENAME>" "eingetippt" wird und dann das C64 KERNAL vom EasyCartridge lädt, als wäre es ein Laufwerk. In der Hardware haben wir das aber schon immer supported. Es gibt da spezielle Signale namens IOE und IOF bei denen man entsprechend reagieren muss.

Again, we are looking for something that worked until autumn/winter 2019 for r2 machines (no later boards did exist at this time). Can we find what happened there to get easyflash working again with the native c64 mode of the MEGA65?

lydon42, 2023-04-25:
sy2002 remarks that the RAM access depends on the IOE and IOF signals.