MEGA65/mega65-core #707: Color parameter in DOT Basic command has some issues

sir-lazarus, 2023-05-24:
**System:**
On Mega65 and XEMU with ROM 920377/920383.
**Bug description:**
When using the color parameter for the DOT Basic command, the pen color will not be set, so ignored, under following circumstances.
**To Reproduce:**
When use the PEN command first to set a color and then execute the DOT command without color parameter then every following DOT command with color parameter ignores the newly setted color.
If you set with PEN command a new color and then use the DOT command with color parameter, this time the colour is set correctly.

**Code to reproduce**
10 SCREEN 320,200,4 :SCNCLR 6
20 PEN 1: PRINT "SET  PEN:";RPEN(0)
30 DOT 10,10  : PRINT"READ PEN:";RPEN(0)
40 DOT 11,11,2: PRINT"READ PEN:";RPEN(0)
45 PEN 0: PRINT"SET  PEN:";RPEN(0)
50 DOT 12,12  : PRINT"READ PEN:";RPEN(0)
60 DOT 13,13,3: PRINT"READ PEN:";RPEN(0)
70 PEN 0: PRINT"SET  PEN:";RPEN(0)
80 DOT 14,14,3: PRINT"READ PEN:";RPEN(0)
100 GETKEY A$
110 SCREEN CLOSE
**Screenshots**
RUN
SET  PEN: 1
READ PEN: 1
READ PEN: 1
SET  PEN: 0
READ PEN: 0
READ PEN: 0
SET  PEN: 0
READ PEN: 3

READY.

[image]

dansanderson, 2023-05-26:
This is probably a ROM issue not a core issue, and is probably the same issue as this: https://github.com/MEGA65/mega65-rom-public/issues/59

lydon42, 2023-05-26:
I agree.

closed 2023-05-26 by lydon42