MEGA65/mega65-core #724: Generate a MAC address silently during on-boarding

dansanderson, 2023-08-28:
A new MEGA65 with a fresh SD card starts with a MAC address of all zeroes. This is a non-functional MAC address. To enable networking, a user must open the Configuration utility, navigate to the Network tab, then press the R key. This key is not mentioned on the screen. (It will be mentioned in the new User Guide.)

An improvement would be for the on-boarding process to silently generate a MAC address and store it with the config. This removes an obscure step in the common case of a user keeping the SD card they used during on-boarding. The "R" keypress would only be needed for users that create a fresh SD card.

Even better: the SD card utility could also do this when initializing config data. Then everyone gets a random MAC address assigned without user interaction.

lgblgblgb, 2023-08-28:
I am really not sure, but IIRC the 8 byte UUID from `$FFD7100` is mentioned that it can be used to generate "unique" MAC address. Surely the MAC address itself should be in a range of assignment to a vendor (btw, like Commodore? starting with `00:80:10` if I remember correctly). Possible even HYPPO can/should check on every "boot" that the configured MAC address from $D6E9 is at least all zero or all $FF and if so, initialize it automatically by setting the first three bytes of the MAC address to 00:80:10, and some value based on the UUID registers. This way, even if HYPPO must "re-generate" MAC, a given MEGA65 would get the same again. And if uses sets another MAC, HYPPO does not detect the "empty MAC address" situation so it will leave it alone, as-is. I can't say it's a good idea anyway, but I thought better to mention still.

lydon42, 2023-08-28:
Configure sets MAC to a random local unicast MAC, and we just should do the same in configure.

We are no official vendor, so we should not some other Vendors address space!

ki-bo, 2024-01-15:
As mentioned, we can't use MAC address vendor IDs that we don't own. For such a use-case we have the local unicast MAC address space (setting bit1 and clearing bit0 of the first byte will ensure this). The random MAC address generator of CONFIGURE already does that, and ONBOARDING should do the same automatically, so a user will always start with a valid MAC address.

Entering ONBOARDING or formatting the SD card will result in a new random MAC address. The user will need to edit it afterwards if a specific one is desired.

closed 2024-01-21 by ki-bo