MEGA65/mega65-core #726: Fail faster when floppy drive is empty

dansanderson, 2023-08-29:
When the internal 3.5" floppy drive is mounted and does not contain a disk, an attempt to access the disk pauses for a few seconds. It would be an improvement to shorten this delay as much as possible.

This is especially true in the case where the MEGA65 is configured to auto-mount the floppy drive, and there is no disk in the drive when the computer is switched on. (Configuration Utility: Chipset > F011 Disk Controller = 3.5" Floppy Drive.) In this case, boot pauses before displaying the READY prompt.

ki-bo, 2024-01-05:
I just tried this again, and for me this is even worse:

`MOUNT
DIR`

will hang forever. I can hit RUN/STOP + RESTORE, then the cursor is gone. But in general, the error handling of the internal drive seems to be broken. This is already the case with a 0.95 release core btw.

lydon42, 2024-01-05:
This is something the to be refurbished d81_detach0/1 will do better, as it will include the option to directly mount the internal drive in the detach call.
See #628

ki-bo, 2024-01-05:
> This is something the to be refurbished d81_detach0/1 will do better, as it will include the option to directly mount the internal drive in the detach call. See #628

I am not sure if I can follow you. This issue is not about problems when switching between images and real floppy drive, but rather the general problem that the ROM locks up if you try to access the drive while there is no disk inserted. This already happens if you don't do `MOUNT` in Basic (or via Freezer) but already if you just set CONFIGURE to select the internal drive on boot-up.

ki-bo, 2024-01-05:
As the original Commodore ROM also locks up, and all MEGA65 ROMs I tried to far, makes me wonder whether the code has an issue behaving weird when reading without any disk inserted.

gardners, 2024-04-27:
@ki-bo and I have determined that the index hole and RDATA lines both are idle when there is no disk inserted. We can thus fail early if the motor is turned on, but no changes are noticed on either of those lines.

gardners, 2024-04-27:
floppy seems to be correctly marked as not present when motor is spinning, but RNF is not asserted immediately when trying to read a sector, when we know that no disk is inserted. As a result, the ROM still waits many seconds when starting up without a floppy in the drive if using the real drive is enabled by default.

gardners, 2024-04-28:
This little program can be used to help investigate:
```
10 POKE$FFD3080,$60                                                              
 20 SLEEP 1                                                                       
 30 POKE$FFD3084,$00                                                              
 40 POKE$FFD3085,$01                                                              
 50 POKE$FFD3086,$01                                                              
 60 POKE$FFD3081,$40 : REM READ SECTOR                                            
 70 PRINTHEX$(PEEK($FFD3082));" ";HEX$(PEEK($FFD3083))                            
 80 IFPEEK($FFD3082)AND$80 THEN SLEEP 0.2: GOTO 70           
 ```

gardners, 2024-04-28:
It finishes quickly with a disk in, but takes ~1.5 seconds without a disk, and the RNF flag (bit 4 of $FFD3082) never gets set until after BUSY is cleared.

gardners, 2024-04-28:
Was not checking disk present flag during reads. This is likely to fix the problem

gardners, 2024-04-28:
Confirmed fixed.

gardners, 2024-05-02:
@ki-bo noticed a bug: Running DIR fails every other time now.
$D082 contains either 4 or 32, with 32 typically meaning it will work, and 4 that it will fail. But if run in a BASIC program loop it works everytime. So I am guessing it is about the timeout for disk detection triggering _after_ the last request has been serviced, and before the computer looks at the drive again.

Holding down F3 so that DIR runs continuously from BASIC prompt still has it fail every other time, though.

lydon42, 2024-05-27:
As this is still not the full fix, I just reverted the changes from development.