MEGA65/mega65-core #733: TI drifts in NTSC mode

dansanderson, 2023-09-14:
**Test Environment (required)**
Platform: MEGA65
ROM Release: 920385
Core Commit: cae3b0b

**Describe the bug**
The TI timer variable counts inaccurately in NTSC mode.

**To Reproduce**
```
5 T=TI
10 ?TI-T:GOTO 10
```
Run this along with a stopwatch. In PAL mode, it matches. In NTSC mode, it drifts by about a second per minute.

**Additional context**
This was originally filed as https://github.com/MEGA65/mega65-rom-public/issues/78 to research whether the ROM is setting registers correctly. We fixed an issue where TOD50 was not being set correctly in NTSC mode, but this didn't fix the drift, and would not have adequately explained the drift anyway. I don't know enough about expected behavior of the CIA to know if this is a core issue, but I don't have any active leads on changes that could be made in the ROM, so I'm creating a corresponding core issue for further research.

lydon42, 2023-12-12:
Something seems to be wrong here... if I change NTSC/PAL with freezer, then do clr ti, I always end up with 193 in DD0E. This seems to be wrong.

lydon42, 2023-12-12:
And it's 50Hz vs. 60Hz, which means if you have the wrong TOD50 setting you should be loosing a just over a second every ten seconds (this is what I was seeing in my 14 sec RTC tick test).

dansanderson, 2023-12-12:
I only see references to CIA2 in the ROM code for TI, I don't know why.

lydon42, 2024-08-17:
The rom issue has been long fixed, so I guess we can close this, too.

closed 2024-08-17 by lydon42

dansanderson, 2024-08-17:
I believe we left this issue open to further investigate an underlying issue in the core with the CIA TOD50. The ROM has been fixed to use a different scaling constant for NTSC mode instead of flipping TOD50, so the BASIC TI is good.

In the other ticket I said, "I tried to make a version of this that set TOD50 based on the VIC_PAL register, and it didn't seem to make a difference. More research is needed to figure out why." It's possible I was just using it incorrectly, but I think I expected flipping TOD50 to fix this issue without an alternate scaling constant. /cc @johnwayner