MEGA65/mega65-core #736: Implement hardware-accelerated cycle-exact IEC bus controller device

gardners, 2023-09-22:
**Is your feature request related to a problem? Please describe.**

#341 and #283 both occur because managing IEC communications on a CPU that changes frequency is a bit of a pain in the backside.  It would be nice to have a good solution to this, so that all IEC peripherals worked -- including in fast/JiffyDOS/DolphinDOS mode etc.

**Describe the solution you'd like**
A hardware accelerated IEC serial controller that looks after all the black magic and exact timings required, and presents a hardware interface with commands that map to the KERNAL jumptable for IEC communicatoins.

**Describe alternatives you've considered**
1. Acquire time-machine, go back to when 6522 was being developed, and point out the shift-register bug, so that the VIC-20 would have used fast serial from the start. This would make a lot of things easier, however, timemachines are never in the depot when you go looking for one, which also presents some interesting logical corrundrums, similar to the paradox of why space isn't as bright as the surface of a sun, given that all sight lines point towards stars.
2. As above, but this time tell them to get over the lack of shift-register, and just use a JiffyDOS-like protocol from the start in the VIC-20.
3. As above, but start a company who makes cheaper IEEE-488 cables, so that the VIC-20 could have just stayed with the parallel bus. Then we could all still be using SFD1001s etc, too.
4. Spend a lot of time and effort in the ROM to make it all work. Volunteers welcome.

**Additional context**
Also has the benefit of making it much easier to access IEC devices without the KERNAL, as the JSRs to the jumptable will instead basically become a set of similar POKEs.

gardners, 2024-01-04:
Standard and JiffyDOS serial protocols are now believed to be working correctly.

There is one work-around, due to a mystery in iec_serial.vhdl where micro_wait() results in double the correct duration of delay when synthesised, compared with under simulation.

See the following blog posts for development log info for this issue:

https://c65gs.blogspot.com/2024/01/hardware-accelerated-iec-serial-bus.html

https://c65gs.blogspot.com/2023/12/hardware-accelerated-iec-serial.html

https://c65gs.blogspot.com/2023/10/hardware-accelerated-iec-serial.html

gardners, 2024-01-04:
1581/1571 fast serial is being worked on, but is going slowly due to lack of a good disassembly of the 1581 or 1571 ROM, that would make debugging things much easier.

dansanderson, 2024-06-03:
Summarizing status to discuss next steps: We started a project to implement an IEC serial/disk interface using VHDL to resolve timing issues that so far have proven impractical to fix in the ROM code alone, and to offer an easy-to-use interface to ML programs.

gardners implemented the VHDL side, and this is believed to be feature complete on the mega65-core:736-hardware-iec-accel branch. Note that another branch named 736-iec-hw-accel exists and should be deleted (carefully so as not to delete the wrong one 🙂 ). The feature complete code should not interfere with existing operation—it doesn't connect to the IEC unless requested—and should be safe to merge to development for further testing.

johnwayner implemented the ROM side. This is partially committed on the mega65-rom:hw-iec branch, with a bit of uncommitted code still on Wayne's hard drive. The last known good state is believed to be working with at least some IEC disk devices, including JiffyDOS support.

Next steps:
1) Delete mega65-core:736-iec-hw-accel branch.
2) Up-merge mega65-core:development to mega65-core:736-hardware-iec-accel. Build and test for regressions.
3) Down-merge mega65-core:736-hardware-iec-accel to mega65-core:development.
4) Up-merge mega65-rom:development to mega65-rom:hw-iec.
5) [johnwayner] Review local changes, test with new core, and push to mega65-rom:hw-iec when ready.

If the core changes appear regression-free, I propose we promote them. We can decide on the rest of the testing and promotion schedule based on how ready the ROM changes are and how close we are to a v0.97 release. For example, we might allow the core changes into the v0.97 release, then wait to promote the ROM changes until the next beta after that, to minimize ROM-core dependency confusion.

commit bdf94745 (2023-12-26): more R3 build fixes #736
commit 79ca4eab (2023-12-26): ... and another R3 build fix #736
commit 50bf7b8e (2024-01-03): try to fix JD RX timing with 2x speed fix #736
commit 61c5b6bc (2024-01-05): reduce time for 1541 UI- again, now that cause is fixed #736
closed 2024-10-22 by lydon42