MEGA65/mega65-core #747: Make RRB even more amazing

gardners, 2023-11-11:
**Is your feature request related to a problem? Please describe.**
Updating RRB lists when you have lots of sprites is a lot of work, and also means you have to dedicate a lot of screen RAM and colour RAM to allow for the longest line, or you have to move stuff around all the time with variable length screen lines.

**Describe the solution you'd like**
Add an RRB token that means "switch to list mode". In this mode, the RRB renderer would treat following tokens as RRB list addresses instead of RRB glyph tokens.  A special End-of-List token would be required in the lists. When such a token is encountered, the RRB renderer will advance the address in the original RRB screen line (that we recall is now being used as a list of RRB lines in this mode). If that token is not an End-of-RRB-Lists token, then it will proceed to render the indicated RRB data. If RRB cycles are exhausted, then it will of course stop, as normal.

The RRB list address tokens should also include the initial X position, Y offset and Y mask information, so that pixies can be drawn as required, without having to change the RRB data as they are scrolled vertically or horizontally.

**Describe alternatives you've considered**
Clearing my backlog of household chores instead.

**Additional context**
This concept is designed to not break backwards compatibility, or require the use of any more VIC-IV registers, as we have pretty much filled up the entire register address space.

gardners, 2023-11-11:
It makes sense to also include an option for relative GOTOX as part of this, so that pixies in separate lists can also have multiple layers of rendering.

deathybrs, 2023-11-11:
At Paul's suggestion, I am adding to this the idea of having GOTOX have both a relative and absolute value for the XPOS, as this would allow for single layer moves at the beginning of the line, but then the rest could be static, allowing us to do a LOT less updating.