MEGA65/mega65-core #756: System files in a subdirectory

dansanderson, 2023-12-14:
**Is your feature request related to a problem? Please describe.**
Currently Hyppo expects to find all system software (.M65 files) and the ROM (.ROM files) in the root of the SD card.

This means there can only be one set of system software per SD card. If I have multiple MEGA65 cores installed, they must all be compatible with a single set of system software. This makes it difficult to do end-to-end testing of releases, and of issues that apply to earlier releases, because mismatched system software doesn't provide a consistent test environment.

There's a partial solution for the ROM specifically: one of several ROMs can be selected by holding a number key during boot. To select an alternate MEGA65 core and matching ROM, I have to remember which slots go with which, and press the number key at the right time after selecting the core.

As a secondary consideration, the SD card root directory appears cluttered from any view that doesn't hide the system files, e.g. on a PC. (The Freezer, and U12 device generally, already hides M65 and ROM files from file lists, so this is not a major concern on device.)

**Describe the solution you'd like**
Hyppo searches for system files and ROMs in three places on the active SD card: `/mega65/slot<N>/` where N is the core slot number, `/mega65/`, then `/`. This allows for a set of system files per MEGA65 core, a default set, and backwards compatible support for the current root dir layout.

Hyppo uses the same search path for alternate ROMs. If I'm booting core slot 2 and hold down the 1 key, it searches for `mega651.rom` in `/mega65/slot2/`, `/mega65/`, then `/`. If not found in any location, Hyppo fails with a message as it does today. Without holding down a key, it searches for `mega65.rom` in those locations.

The SD Card utility installs core-embedded system files to `/mega65/slot<N>/` by default, maybe with a prompt to offer `/mega65/` or `/` as alternatives. As a new feature, it also offers the ability to install core-embedded system files this way even when not formatting a new card, i.e. as a step after installing an alternate MEGA65 core.

**Describe alternatives you've considered**
A stricter version of this would only offer `/mega65/slot<N>/` for MEGA65 cores going forward. Older MEGA65 cores would use `/` on their own anyway.

Another option is to reverse the lookup order: `/`, `/mega65/`, then `/mega65/slot<N>/`. This would allow the user to override a system file or ROM for all MEGA65 cores without having to copy the same files into multiple directories. It's not what I would want as a tester—I'd prefer multiple self-consistent environments, at the cost of managing multiple file sets in directories—but maybe this would be more intuitive for people that like to mess with ROM betas or HICKUP.M65 files or whatever.

Another option: name the subdirectory after the core Git commit ID, e.g.: `/mega65/development,20231123.08,4259566/` (or just `/mega65/4259566`) This would allow a given core version to find its files regardless of which slot it's in, at the expense of making it more difficult for users to manage files manually. But maybe this is better than having the user remember which slot contains which core.

lydon42, 2023-12-14:
The startup  slot is unknown to the core, so using the slot number will be not easy. The commit hash would be easier to source, as it is even accessible through a register.
Searching will make thinks slower in startup. One main goal was to have a very fast startup, and not do to many things that would make booting less like the "retro" style instant on.

dansanderson, 2023-12-15:
I wonder if this feature request is mostly of interest to testers, and therefore could be a secondary mode of operation that's OK to slow down startup (by some amount). Keep `/` as the first check (presumably the same speed), and only check `/mega65/<versionid>/` if system software files required by the core are missing from `/` before halting with an error message.

lydon42, 2024-09-01:
The first load that is done is `BANNER.M65` followed by `HICKUP.M65`.

This could mean that we catch the failure to load either of them (why is the banner loaded from card? we do put it into the core at the end...), and then go to search for the subdirectory and store it's directory cluster for lookup.

Then this will get tricky, as it changes directory in therefor interferes with disk image load. So for this to work the whole subdirectory topic needs to be implemented into the base of HYPPO.

Tagging #765

lydon42, 2024-09-07:
We agree that this is an interesting feature for testing setups with multiple core branches running, but it can't be implemented unless HYPPO DOS and process handling knows about paths.

lydon42, 2024-09-08:
Idea how SLOT NUMBER based subdirs could work:
- add a core header field SLOT NUMBER
- fill this field in MEGAFLASH when flashing a CORE to a slot
- use this field to select `/M65/SYS/S[0-9]/` as the path