MEGA65/mega65-core #757: Thumbnails are blank in freezer

gardners, 2023-12-17:
**Test Environment (required)**
You can use [MEGA65INFO](https://files.mega65.org?id=3eeb7d82-8bd7-44b7-82b6-cef29d123b0e) to retrieve this.
 - Platform: (DevKit, MEGA65, NexysA7, xemu)
 - Core Commit: (7 digit hex number, press MEGA-TAB)
 - ROM Release: (i.e. 920XXX)

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Thumbnails visible as before

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**

From @Johnwayner in discord:
It appears that the addressing changes in https://builder.mega65.org/job/mega65-core/job/development/17/ broke the ability of the freeze code to read from the thumbnail generator addresses at $FFD2640 and $FFD2641.  Thus the freezer is just reading some memory somewhere else.  My VHDL is very weak.  <@824166878284349461>  might be able to help with this as he was involved with these changes.

This will narrow it down quite a bit.

gardners, 2023-12-17:
Confirmed that mega65-tools/src/tests/thumbnail.a65 still displays the thumbnail.
It is only the freezing process that doesn't work.

thumbnail.a65 uses $D640 in M65 IO memory map, i.e., effectively using $FFD3640.

HYPPO is currently uses $FFD2640.

That _should_ be ok, though, as $FFD2640 should also have the thumbnail mapped there.

Just confirmed that $FFD2640 also works, by reading it from BASIC65.

So the problem seems to be it not being accessible from Hypervisor memory context.

dansanderson, 2024-07-01:
I believe this is fully fixed as of release v0.96. This can be closed.

closed 2024-07-02 by lydon42