MEGA65/mega65-core #767: Errors with physical floppy when no D81 image gets automounted on boot

KiDraDragon, 2024-01-05:
**Test Environment (required)**
You can use [MEGA65INFO](https://files.mega65.org?id=3eeb7d82-8bd7-44b7-82b6-cef29d123b0e) to retrieve this.
 - Platform: (DevKit, MEGA65, NexysA7, xemu)
 - Core Commit: 93D55F0, 93E2EB8, 7FB8B6F
 - ROM Release: 920377, 920391

**Describe the bug**
When no D81 image gets automounted on boot (e.g. MEGA65.D81 is not present on the SD-Card), then there are errors with the physical floppy drive. When cold booting the MEGA65 and doing `MOUNT`, then `HEADER"DDFLOPPY",I01` DOS throws `75,FORMAT  ERROR,40,00`. Additionally, when cold booting the MEGA65 and doing `DIR` a `73,MEGA65 CBDOS 920391,00,00` appears in red after the listing. When doing a cold boot, then `MOUNT`, `DIR` or `LOAD"$",8`and then `HEADER"DDFLOPPY",I01`, it works.

When a MEGA65.D81 image is present and gets automounted, the red message after the directory listing and the error after HEADER, do not appear and everything works.

**To Reproduce**
Steps to reproduce the behavior:
For HEADER error:
1. Cold boot MEGA65 without any D81 automounting.
2. Execute `MOUNT`with a physical floppy inserted
3. Execute `HEADER"DDFLOPPY",I01`

For red message after directory listing:
1. Cold boot MEGA65 without any D81 automounting.
2. Insert physical floppy into drive.
3. Execute `MOUNT`
4. Execute `DIR`

**Expected behavior**
HEADER should just work and after DIR no red message should appear, no matter if a floppy image gets automounted on boot or not.

**Screenshots**
[image]
[image]
[image]

**Additional context**
I tested this with the original physical drive that the MEGA65 shipped with and another, new-old-stock floppy of the same type (which works better in general).

johnwayner, 2024-01-05:
FYI, this is a ROM issue -- not a core one.  Basic wasn't fully mounting the internal drive.

lydon42, 2024-01-05:
Not a core issue, and #760 will address better HYPPO interface so ROM needs not to worry about mounting the physical drive it self.

closed 2024-01-05 by lydon42