MEGA65/mega65-core #775: Core 0.96 test: 1351 mouse is jittery in applications like MegaMOD and Sprite Editor in Freezer

MauriceMEGA65, 2024-01-22:
**Test Environment (required)**
MEGA65 Release 0.96 "Batch 3" Release Candidate
 - Core Commit: 68aabb0
 - ROM Release: 920392
 - R3A (batch1) hardware
 - Mouse is connected to port 1, and set as Normal Mode in config.

**Describe the bug**
1351 mouse is jittery in applications like MegaMOD and Sprite Editor in Freezer, but seems OK in BASIC

**Expected behavior**
Stable mouse pointer, like when using an Amiga mouse

https://github.com/MEGA65/mega65-core/assets/75681098/11d42d48-2e66-400c-acd2-34618b8305dd

https://github.com/MEGA65/mega65-core/assets/75681098/640cfed1-3e62-41b7-a375-a58f40bc981b

https://github.com/MEGA65/mega65-core/assets/75681098/c150ccf1-0c3e-40b0-be11-0a7b5e6a57d8

**Screenshots**
See attached videos of MegaMOD, Sprite Editor and BASIC

**Additional context**
https://discord.com/channels/719326990221574164/1196590643007463459/1198582748340834304

gardners, 2024-01-27:
I've been thinking about how to fix this at the hardware level by adding hysteresis to 1351 read values. One option is to fix this in the 751- issue branch, while I have mouse_input.vhdl open for surgery. See #751

gardners, 2024-01-28:
That helps, but the jitter is actually over a range of 3 values, not 2.

gardners, 2024-01-28:
Meanwhile, I filed an issue on the MouSTer repo asking for optional pull-up on Amiga right and middle buttons, that would allow C64s and MEGA65s to read those buttons without requiring a special cable with pullup in it.

gardners, 2024-01-28:
Confirmed this fixes 1351 mouse jitter, provided that amiga mouse aggressive mode is turned off (which now also controls 1351 de-jittering at hardware level).

lydon42, 2024-08-09:
Looking at the commits, this has already gone into release-0.96.0

closed 2024-08-09 by lydon42