MEGA65/mega65-core #779: Amiga mouse in port 1 interferes with keyboard

dansanderson, 2024-01-27:
**Test Environment (required)**
You can use [MEGA65INFO](https://files.mega65.org?id=3eeb7d82-8bd7-44b7-82b6-cef29d123b0e) to retrieve this.
 - Platform: MEGA65 R3 or R5
 - Core Commit: 683-cartflash,20240124.23,65b7373
 - ROM Release: 920377, 920394

**Describe the bug**
ROM 920377: When an Amiga mouse is connected to port 1 and the port is in Amiga mouse mode (aggressive or normal), moving the mouse causes crazy typing events.

ROM 920394: When an Amiga mouse is connected to port 1 and the port is in Amiga mouse mode (aggressive or normal), Run/Stop does not always interrupt a program.

**To Reproduce**
1. Connect an Amiga mouse to port 1.
2. In the Configuration menu, set port 1 to Amiga mode (aggressive or normal).
3. `10 PRINT "HI":GOTO 10`
4. `RUN`
5. Press Run/Stop.

Run/Stop should reliably stop the program the first time it is pressed. What's actually happening is sometimes Run/Stop doesn't do anything.

Disconnect the mouse and try again. Connect a 1351 mouse and try again. Run/Stop works without interference.

(If using a mouSter and wireless mouse, be sure to jiggle the mouse to make sure it is active. Wireless mice will intentionally disconnect when not moved to save battery.)

**Additional context**
It is likely that the symptoms are more mild with ROM 920394 because the new ROM keyboard scanner uses the hardware typing event queue instead of CIA lines for most things, but still uses CIA lines for things like detecting Run/Stop. This strongly suggests that an Amiga mouse on port 1 is simply interfering with CIA1 DC00 return values that overlap with CIA keyboard scanning.

Presumably there should be a way for the Amiga mouse driver to translate to 1351 signals more accurately, so as to not cause these issues.

dansanderson, 2024-01-27:
Not a regression, so not recommended for prioritization with the v0.96 release.

dansanderson, 2024-01-27:
I'm not able to reproduce this with the v0.95 core, so this is a regression.

dansanderson, 2024-01-27:
Paul says this is a fix, will verify: https://github.com/MEGA65/mega65-core/commit/226eebaec9b4b870bc94823173815ca83158a510

gardners, 2024-01-27:
See #751 for work on fixing Amiga mouse interference with keyboard etc.

lgblgblgb, 2024-01-29:
Not sure if it has anything to do with this issue: but during the weekend I had issues that freezer didn't reacted at all to any key, also the flasher the same way but _only_ in the core file selection part (it works before that step). Strangely enough I could type normally without any issue in ROM, both new (using hw accel kbd) and old, plus using a very old and new core as well (for the new ROM, obviously only a new core I could tested with). All my issues went away when I unplugged Amiga mouse from port 2 (not one though!), which seemed to cause all of these problems.

lydon42, 2024-01-29:
Do we need this issue open or can we close this because, as Paul wrote, work is being done in #751 ?

dansanderson, 2024-01-29:
They are distinct issues, so I would opt for keeping them both open, even if the same effort will fix both. I don't know for sure that keyboard interference and right mouse button flicker have the same cause, and a fix might not completely handle both.

lydon42, 2024-08-15:
Is there still a problem or not?