MEGA65/mega65-core #788: CONFIGURE: POWER-CYCLE screen has color issue on VGA only

markkrueg, 2024-02-13:
**Test Environment (required)**
[image]

**Describe the bug**
When exiting CONFIGURE you see the new-ish "POWER-CYCLE" screen. This is showing up very oddly -- almost unreadable blue on blue colors -- on VGA displays. Here is my MEGA65 with DVI display on left and VGA on right.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Hold ALT at boot
2. Choose 1. CONFIGURE MEGA65
3. Complete configuration and save.
4. See "POWER-CYCLE" screen as shown in my screenshot.

**Expected behavior**
"POWER-CYCLE" screen colors should be similar on DVI and VGA outputs and should be readable.

lydon42, 2024-02-14:
Yeah, I can confirm, for me the HDMI switches off and VGA gets a weird color.

M3wP, 2024-02-24:
Hmm...  I'm looking at this and from what I can see, there is nothing in configure that should be causing it.  At the moment, I believe that the problem probably lies in the hyppo save logic.  To confirm this I need to know if the same issue arises when you simply select Done | Exit Without Saving | OK?

lydon42, 2024-04-09:
My observations:
- startup in NTSC mode
- change nothing in configure
- select SAVE & EXIT
- screen goes blank, then switches to PAL
- displays "POWER CYCLE"

This is obviously a error, if I change nothing, nothing should change with my display. And it shouldn't change settings anyway, as the power cycle will take care of loading the new configuration.

M3wP, 2024-04-10:
Updated configure to 1.04.

[mega65_config_1_04.zip](https://github.com/MEGA65/mega65-core/files/14926914/mega65_config_1_04.zip)

I am not getting the issue with this version and the batch release core.  Please advise if I need to test a different scenario or if this update addresses the issue.

M3wP, 2024-04-10:
[mega65_config_1_04.zip](https://github.com/MEGA65/mega65-core/files/14926982/mega65_config_1_04.zip)

I believe this update should address this issue.  Please advise if this is the case.

M3wP, 2024-04-10:
Oh!  I should note that the bug causing the issue is actually in Hyppo in the syspart_trap syspart_configsector_apply_trap.  If this trap is expected to function correctly, it will need be bug-checked.

lydon42, 2024-05-25:
The hyppo part looks correct to me. The problem is the data that is copied to DE00.

It seems that hypervisorApplyConfig calls copySessionOptionsToSectorBuffer, which copies from optSessBase=$C200 to DE00. But if I set a breakpoint and inspect C200, I can only see FF in the first 16 bytes. This explains why the apply will set D06F to Cx instead of 8x, which results in the problem.

Checking C200 directly at the start of configure also shows the wrong values. One would assume that the optSessBase will be the values that one can see in the display of configure, but this seems not to be the case.

lydon42, 2024-05-25:
It seems optSessBase is no really set by anything and contains garbage. M3wP agrees that this might be an artifact from the past. Changing optSessBase to point at the same as optDfltBase will also fix the problem.
```
optDfltBase	=	$C000
;optSessBase	=	$C200
optSessBase     =       optDfltBase
```

There is **no problem in HYPPO**.

I will apply the change that disables apply now, as it is no longer needed. I will also add the optSessBase change with a comment explaining why, so that we remember in the future.

lydon42, 2024-05-25:
latest development core has the fix.

closed 2024-05-25 by lydon42