MEGA65/mega65-core #797: Set GotoX Rowmask switches between bitmap/alternative palette

archiecrux, 2024-03-31:
Choosing the colour palette for rendering the RRB object in advance and determine the rowmask with another gotox results in the colour palette used being changed, probably by using the same value. The only way around this is to forego the alternative palette, or I missed something.

Core: release-0.96,20240224.00,3c10488
Hardware: MEGA65 R3a

CRAM dump:
1460 146F 146F 146F 146F 146F 146F 146F
146F 146F 146F 146F 146F 146F 146F 146F
146F 146F 146F 146F 146F 146F 146F 146F
146F D060 081F 081F 081F 081F 081F 081F
081F 081F 081F 081F 081F 081F 081F 081F
081F 081F 081F 081F 081F 081F 081F 081F
081F 081F 1000 98C0 080F 0C0F

[image]

archiecrux, 2024-08-29:
There was recently a conversation on discord about the method for switching to the alternative palette.
[https://discord.com/channels/719326990221574164/977076334297743360/1260056621321683075]

The previous method (colour ram byte 1 bits 5+6 set if GOTOX bit =1 and ROWMASK bit =0) should be replaced. Bit 5 of colour ram byte 0 should change the palette used in the future.

This issue would then be resolved and can then be closed.

lydon42, 2024-09-01:
Replacing a current method sounds like a incompability.