MEGA65/mega65-core #803: Background D020 updates late after raster interrupt?

dansanderson, 2024-05-31:
A simple assembly language program that sets a raster interrupt for line 200, changes the border (D020) and background (D021) to black (0), counts down from 255 in a short busy-loop, then changes the border and background to white (1) demonstrates that the background does not change until just more than half way across the raster line, for both transitions. The border appears to transition on the first of the two raster lines. This is not explained by CPU cycles spent in the interrupt handler code, at 40 MHz.

I don't actually know if this is unexpected behavior for a C65 or a C64. Is there a known trick to this? Note that at 40 MHz there is no visible "stability" issue often discussed in the context of a C64 1 MHz interrupt.

[image]

[raster_hwirq.asm.txt](https://github.com/user-attachments/files/15507042/raster_hwirq.asm.txt)

dansanderson, 2024-05-31:
I can get the background change position to flicker steadily by eight-ish pixels (just eyeballing) with minor code changes. It would be surprising if this is CPU cycle slop in the interrupt, and seems more likely to be related to when the VIC recognizes a new border color.

I also tried a version where I cycled through 16 colors across 16 adjacent lines, one interrupt per line, and saw an interesting pattern of variability, stable by line position. So that might also suggest the background color change happens at different times based on raster vertical position.

ki-bo, 2024-09-18:
This is probably related to MEGA65 and no C65 behavior. My feeling is that you see pre-rendered characters where the bg color was still white when the rendering happened (chars are not rendered in real-time, but are rendered into the raster rewrite buffer first, and then those pre-rendered pixels are pushed to the video output later). Rendering of chars probably starts at the end of the previous raster line, and parts of the text are already ready when you change the bg color.

ki-bo, 2025-09-17:
Can we close this one? I don't see unintended behavior, the VIV behaves as expected.

dansanderson, 2025-10-16:
The test program attached to this issue (raster_hwirq.asm.txt) produces the same screenshot also attached to the issue above, with core release-0.97,20250422.09,aaf4542.

I think my original concern was that the raster interrupt was not happening when I thought it should be happening. I understand your explanation as, it is happening at the right time, but changing the background color register near this moment doesn't do what I expected it to, because some amount of the line—specifically in the character graphics area—was rendered prior to the interrupt. So the border appears to change on the correct line, but the background appears to change mid-line, with 8 pixels of flicker because that's a pre-rendered character by some amount of timing. All correct?

I wonder how a timing-sensitive graphics program is meant to account for this. But if everyone is comfortable with the way this works, then I'm fine resolving the issue.