MEGA65/mega65-core #811: Feature Request:  AudioDMA IRQs in order to implement advanced buffering

M3wP, 2024-07-18:
The implementation of IRQ generation in the AudioDMA would make it easier to implement double-buffering type scenarios.  This will allow the use of digital audio in a wider variety of applications since the critical fast memory footprint of the media is significantly reduced.

Another method of implementing these buffering techniques is by the use of polling.  This method would be far less accurate leading to audio artefacts and require much more CPU time and more complex timing generation to perform with any reliability.

It is expected that when using IRQs, the buffer size for each channel could be as little as 2 to 4kB.  This is based on slow RAM DMA transfer testing.

The IRQs should be generated per channel based on a point in the playback stream.

There is limited IO space for the feature so a single byte (for "M" or next most significant byte) is all that can be provided per channel.  Plus another byte for the IRQ flags and feature enabling.

The single data byte should be used as an offset from the end point such that when 0, the IRQ is generated when the playback reaches the end point and when another value, when the playback is less that or equal to that distance from the end point.

The IO should be:
    $D713:  IRQ flags and enabling flags
    $D714-$D717: IRQ offset positions

M3wP, 2024-07-18:
I have implemented the functionality in https://github.com/MEGA65/mega65-core/compare/development...M3wP:mega65-core:811-adma-irq

A test and demonstration program.  Run from C64 mode.  Press space to start/stop.

[probe.zip](https://github.com/user-attachments/files/16284706/probe.zip)

lydon42, 2024-07-21:
merged into https://github.com/MEGA65/mega65-core/tree/811-adma-irq for auto build and testing.

lydon42, 2024-08-05:
Tested with MegaInvaders and MEGApple, they still work fine. Will merge this soon.