MEGA65/mega65-core #815: Sprites not correctly hidden by border

ki-bo, 2024-07-28:
Tested with development core d73d8712

All sprites should only start drawing between left and right border. However, they start drawing one H640 pixel too early (into the left border) and stop drawing one H640 pixel too early (last pixel drawn is two pixels before right border start instead of the pixel directly before the border).

This only happens and is visible if scan-lines are disabled and an even SIDEBDRWD value is configured. Setting an odd value there also fixes it. Unfortunately, the default value is even ($50) ;-)

There are two problems:
1) The signal provided to the sprite modules whether the border is currently active or not is only updated with cpuclock (which only as a rising edge every two H640 pixels). It should be updated in the domain of the pixelclock instead. This will fix the difference between odd and even SIDEBDRWD settings behaving differently. However, this will only make the issue consistently appear for every setting of SIDEBDRWD :-). Therefore, we also need:
2) The signal border_in needs to be delayed by one additional cycle before being provided to the sprite module. This will correct the position where the sprite starts and stops rendering relative to the side border.

lydon42, 2024-08-05:
merged into development.

closed 2024-08-05 by lydon42

ki-bo, 2024-09-16:
This seems to be the same issue as #579. There is a unit test for #579, but the code looks bloated and does not work at all. I will derive a new unit test for this one here (#815) to replace the one of #579.,