MEGA65/mega65-core #821: CIA Oneshot timer handled incorrectly

lydon42, 2024-08-20:
It seems that in oneshot mode the timer is not reset to the latched value but stays at ffff.

AmokPhaze101 found this:
that's what should indeed happen according to Magic Disk 64 CIA course : Bit3 (ONE-SHOT/CONTINOUS)
- If this bit is deleted, the timer is in CONTINOUS mode. This means that the counter counts down to 0, is reloaded with the value in the latch and starts counting again.
- If the bit is set, the ONE-SHOT mode is active - it counts down to 0 and is reloaded, but the timer now stops automatically until it is restarted with bit0 of the control register. (edited)

@gurcei is working on it already in cia_oneshot branch ([see discord](https://discord.com/channels/719326990221574164/782757495180361778/1275314691522035753)).

lydon42, 2024-08-20:
Note to myself: squash merge the branch so the issue tag can be added.

Rhialto, 2024-08-20:
My proposal would again be: copy as much as possible from the C64 core, which is known to be pretty good. 

What makes this less than 100% trivial is the fact that apparently register access happens here at full speed while at least the timers run at original ~1 MHz speed (which is bound to produce a difference in behaviour), and the extra hypervisor-only registers.

I looked at doing it myself and had both versions in editors next to each other, but unfortunately the C64 version is Verlilog, not VHDL, and it's structured differently. If I had more VHDL experience I would be more likely to "see through" these difference, but I don't.

gurcei, 2024-08-21:
I probably won't have the appetite (or the skillset as yet) to port verilog to vhdl, still have a lot to learn in both, but if there are others out there that want to do such a port across from c64-core, it's fine by me.

Till then, I'm content with any little wins we can have by addressing little issues like this as we find them.

gurcei, 2024-08-21:
I tried testing various programs on the intro disks to see if my cia-oneshot fix broke anything. I tried a few demos, games and tools, seems to behave ok, I reckon it's fine to merge in.

lydon42, 2024-08-21:
merged cia_oneshot branch.