MEGA65/mega65-core #824: Keys No Scroll/Tab/Alt/Help/F9 also read as joystick port 1 input

dillof, 2024-08-22:
Tested on an R6 mainboard with the latest official release 0.96 3c10488, the keys from the last keyboard column are also read as joystick port 1 input. This is both in BASIC and when directly reading from the CIA. 

This does not happen on an R3 mainboard, at least with core version 0.95 93d55f0.

You can easily test it with the following BASIC program by pressing any of the affected keys:

```basic
10 PRINT"{clear}"
20 PRINT"{home}";JOY(1);"  ",JOY(2);"  "
30 GOTO 20
```

Since it also happens when reading the CIA ports directly, I think it's a core issue, not a ROM issue.

ki-bo, 2024-09-01:
I was running into the exact same issue when implementing MEGASPUTM for Maniac Mansion. As on the C64, the CIA port bits are used for both keyboard scanning and for joystick input. If you really want to be sure that you only see joystick port lines pulled to low, all keyboard matrix lines need to be pulled HIGH during that scan. That way, a CIA input of LOW will really result from a joystick port input. The problem now is that the C65 had more keys than a C64, so the 8 bits for the matrix scan were not sufficient anymore. Commodore therefore decided to use 9 bits for the matrix scan. The 9th bit is used from port E of the C65 UART: You need to also control bit 1 of port E. $d607 is the data port and $d608 is the DDR. Once you make sure to set that port bit to OUTPUT and HIGH during the scan, the additional keys in the keyboard matrix will not be able to produce a LOW input anymore (which will provide a mismatch with joystick input).

Please be aware that switching the outputs between LOW and HIGH will take some latency to take effect, to don't just do that immediately before reading the inputs. Be sure to provide some time to let those CIA configuration settle.

ki-bo, 2024-09-01:
Maybe re-test your example with the following amendment:

`15 POKE$D607,2:POKE$D608,2`

This will set bit 1 of UART port E to HIGH and OUTPUT, and you should not see any mismatch anymore. Maybe the JOY scan in ROM should be adapted to have this bit pin controlled during the scan. But the hardware part seems to work as documented.

dillof, 2024-09-01:
Yes, that works, thanks.

dansanderson, 2024-09-01:
I have created corresponding issues for the ROM and documentation, so we can fully describe what a joystick controller needs to do at some point.

I gather that there are no changes needed to the core for this, so I'll resolve this issue. Please re-open if I'm missing a need for a core change.

closed 2024-09-01 by dansanderson