MEGA65/mega65-core #826: DMA src/dst addresses wrap around at MB boundaries

ki-bo, 2024-08-29:
Tested with 0.96 release core

A DMA job filling or copying data does not update its configured MB part of the src/dst address during job execution. For example, if a copy job crosses a MB boundary for its src address from $81fffff to $8200000 it will actually jump back to $8100000.

lgblgblgb, 2024-08-30:
Is this a bug? I remember when I wrote the emulation of this, and Paul said, this is the desired behaviour to be wrapped within the 1Mbyte region, ie DMA cannot cross 1 Mbyte boundary by design. Well, "IIRC" ...

lydon42, 2024-08-30:
Good point @lgblgblgb 
@gardners also mentioned that this is a protection so that a DMA job does not cross from  $FExxxxx to $FFxxxxx.
So perhaps we should keep this as it is, as protecting against this overflow is a rather special check and would effect other address ranges, like $BExxxxx to $BFxxxxx (end of 64MB Attic RAM) or $8Exxxxx to $8Fxxxxx (end of 16MB Cellar RAM).

lgblgblgb, 2024-08-30:
Also there can be corner cases when something other depends on it, ie, doing a DMA job at the end of an 1Mbyte "slice" (right at the end) with chained jobs, so if MB is updated the next DMA "sub-job" in the chain will experience different MB setting which was not the case before.

ki-bo, 2024-08-30:
After discussion with @gardners we propose to add another DMA option to disable the 1MB limit for the current job.
- This will not introduce backwards compatibility issues with existing code
- That option will be ignored in HYPPO context to not impact security

lydon42, 2024-09-04:
Moving this to near future, is it no release requirement.
Steering Group has approved the implementation as a DMA option.

lydon42, 2024-09-09:
Steering Group decided to include the global option bit (4 votes for global bit plus job option, 2 votes for job option only, 2 abstains).

So please also implement the global option bit in $D703.

closed 2024-09-14 by gardners