MEGA65/mega65-core #831: SID: Enhance or reimplement SID

lydon42, 2024-09-10:
The never ending quest to get a better sounding SID implementation. Who will take the crown?

This is a continuation of #449

dansanderson, 2025-04-23:
This Epic needs specific bugs to be actionable. Obviously all Epics do, but I want to call this out here, because we still have an unclear path to 1.0 for the SIDs.

@MassiveBattlebotsFan has said the only remaining to-do for the 8580 is work on the filter table. Is this accurate? Should we have a "fix 8580 filter table" bug, or do we need to break that down into more specific items? Is the work involved to analyze an 8580's filter table, or just massage the existing table until some examples sound accurate, or some other way to derive the correct filter table?

MassiveBattlebotsFan, 2025-04-28:
@dansanderson the filters need to be rewritten, the current ones are effectively beyond repair as far as I can tell. Most of everything else has been cleaned up.