MEGA65/mega65-core #849: Iron out ghdl and ghdl-frame-gen related things

gurcei, 2024-12-25:
Just wanted a ticket to pool some efforts I had made towards getting ghdl and ghdl-frame-gen tool to run better.

I might throw in some extra things to help windows/cygwin users of this repo too (so they don't get issues with EOL differences between linux and windows).

lydon42, 2025-01-11:
@gurcei You are replacing the whole Makefile in your commit, because you change indention. This makes it had to see the actual changes.
Please revert that and make it in two parts: first replace only indention, then do the changes required for windows. Otherwise nobody can check this.

gurcei, 2025-01-12:
Okies, that's something I can aim for whenever I get back to exploring ghdl. The aim of this branch for me was to preserve efforts made so far in relation to this endeavour so that I wouldn't lose those efforts, not to put it up for review at this stage, it isn't ready for that. I can focus on cleanup matters when the branch is more ready for review.

lydon42, 2025-01-12:
I just wanted to put that out before a lot of additional changes are made and retroactive changing would be hard.

If the whole commit is just indention and/or line endings, I do have no problem with it, it is just not easy to see and the comment does not state this clearly.