MEGA65/mega65-core #857: [nexys4ddr-widget] Recent development seems to be stuck in ROM startup

lydon42, 2025-01-16:
The recent development builds of mega65-core (at least140, perhaps also 138 and up) seem to be stuck in ROM startup (blue screen, no prompt).

We need to find out which commit did break things for nexys.

lgblgblgb, 2025-01-16:
Maybe I'm totally off now, but this reminds me a very similar Xemu issue back to then: https://github.com/lgblgblgb/xemu/issues/395

However that was a CIA emulation problem/inaccuracy in Xemu which caused that: I wouldn't think this is the case here, that CIA implementation in the core breaks only with the Nexys board but not otherwise. Anyway, I guess I add this comment just in case. If it doesn't help, feel free to delete this comment of mine to reduce the noise :)

geir-straume, 2025-01-17:
On my Nexys A7, development build 132 is the latest that works. This is with ROM V920395.
Builds 134-141 just get stuck on the blue screen. (I could not find any build 133.)

lydon42, 2025-03-09:
It seems the thing is stuck in reset, I see it going through FA16 again and again (which is the address stored in ($FFFE). So there must be something in the reset process that does not work out as it should.
Testing with 920408.
Could it be something from the IEC branch that does not work out anymore, as the nexys does not have a IEC port?

0xa000, 2025-03-11:
On the QMTECH Wukong board, a core build from commit 347143a (Mar 9) also hangs on a blue screen without a prompt.

0xa000, 2025-03-11:
Ok, it seems I've found a one line fix.

0xa000, 2025-03-11:
I suppose it would be a good idea to check if all the other IEC related signals behave as a IEC port with nothing connected to it when not generating the hardware IEC component. Otherwise the DOS might still run into issues, I guess. However, I've tried mounting a D81 and loading a PRG from it, and that at least seems to work fine.

closed 2025-03-16 by lydon42