MEGA65/mega65-core #864: Full FORMAT of D81 disk image hangs; DIR of full FORMATted internal floppy hangs

dansanderson, 2025-02-08:
**Test Environment (required)**
You can use [MEGA65INFO](https://files.mega65.org?id=3eeb7d82-8bd7-44b7-82b6-cef29d123b0e) to retrieve this.
 - Platform: MEGA65R6
 - Core Commit: Release v0.96 3c10488
 - ROM Release: Release v0.96 920395

**Describe the bug**
A full format of a D81 disk image using the FORMAT command with an ID stalls.

A full format of an internal floppy disk makes OK sounds and ends after an appropriate duration, but a DIR on the result stalls.

**To Reproduce**
1. Freezer, drive 0, create a new disk image. Resume.
2. `DIR` to confirm disk image is working.
3. `FORMAT "NEWNAME",IXX`

Expected: should finish within a minute.
Actual: Stalls in CBDOS code.

1. Insert an expendable 3-1/2" DD floppy disk in the internal drive.
2. `MOUNT`
3. `FORMAT "NEWNAME",IXX`
4. Format finishes.
5. `DIR`

Expected: displays directory of new empty disk.
Actual: Stalls.

**Additional context**
I have confirmed that I'm using v0.96 core and ROM. I need help confirming whether this is true of all R6 machines running v0.96 stable, and whether it also applies to R3. I was inspired to test because someone else also reported a stall in full formatting.

Quick formatting (without the `I` parameter) is working.

dansanderson, 2025-02-08:
We have one report of no reproduction on an R3 board. It is unlikely that R3 and R6 will differ in this way, so I'm still looking for other causes.

I confirmed that a D81 disk image created with something other than the Freezer has the issue.

I can repro using both my external microSD card and my internal SD card.

lgblgblgb, 2025-02-08:
I've tested my setup here, also hangs.

Board: R3, ATRIX version: B8756F4A according to HELP key in FREEZER (btw, I need a reset, since busy bit being stuck made not even freezer could load/come up)

ROM v920410 according to matrix monitor execution is somewhere around 97A1,97A3,979E and such values in general, if it helps. It's often at a place "BIT $D082" which is the FDC status register IIRC, so probably it waits for the operation to finish? Again, according to matrix monitor I/O reg D082 (FFD3082) has the value $A1 at that point, so bit 7 is set, indicating busy.