MEGA65/mega65-core #865: MAKEDISK freezer utility creates technically invalid images

M3wP, 2025-02-08:
**Test Environment (required)**
 - Platform: current
 - Core Commit: bc42a72
 - ROM Release: not applicable

**Describe the bug**
The MAKEDISK utility does not correctly format the DOS header block at track 40, sector 0.  There needs to be a DOS version and correctly space padded label and id.  Further, the DOS type should be "3D".

**To Reproduce**
Steps to reproduce the behaviour:
1. Freeze
2. Pres '0'
3. Select new image
4. Type a name
5. Analyse the created image with a sufficient tool
. 

**Expected behaviour**
The disk image should be a valid DOS disk.

**Screenshots**
Current result (error):  

[image]

Expected result:

[image]

**Additional context**

M3wP, 2025-02-08:
I have a fix for this

lydon42, 2025-03-15:
This is a freezer issue. See [freezer PR#96](https://github.com/MEGA65/mega65-freezemenu/pull/96).

closed 2025-03-15 by lydon42