MEGA65/mega65-core #867: HYYPO Read Dir is returning type bits for files with long file name incorrectly

M3wP, 2025-02-13:
**Test Environment (required)**
You can use MEGAINFO from within FREEZER to obtain this information
(long press RESTORE to enter FREEZER, then press HELP to start MEGAINFO)
 - Platform: any
 - Core Commit: bc42a72
 - ROM Release: any

**Describe the bug**
When using HYPPO Read Dir to get file entries, when encountering a file with an "LFN", it is returning the filetype bits in its entry from this LFN detail instead of from the 8.3 details.

**To Reproduce**
Steps to reproduce the behaviour:
1. Use HYPPO Read Dir, consult the manual.  its a significant process
2. test the bits

**Expected behaviour**
The file type bits should not be incorrect.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

lydon42, 2025-02-13:
First evaluation: I don't think this is true, as FREEZER does not have problems with LFN dirs nor files. And this is also using HYPPO readdir and is checking the type flags.

M3wP, 2025-02-13:
What is it checking for SYSTEM and HIDDEN?  If not, it isn't detecting the problem.  If it is only checking DIRECTORY then all will be well for it.  I am getting SYTEM and HIDDEN for files which I shouldn't.

lydon42, 2025-02-13:
If this would be true, then FREEZER would not be able to find LFN directories. This is not the case. So this is still no bug to me, until you can provide a testcase I can run and that confirms the problem.

LFN entries are always 0x0f in the attributes. Only the following SFN entry will have attributes set to 0x10 to indicate a directory. FREEZER diskchooser does find that, therefor HYPPO works correctly.

lydon42, 2025-02-13:
Not a bug.

closed 2025-02-13 by lydon42

M3wP, 2025-03-10:
The documentation is misleading and/or incorrect.  In one column it states one thing and in another, something else.  The documentation states that there are no NULLs.  This not correct as there are in fact TWO NULLs following the 8/3 filename.