MEGA65/mega65-core #872: Drop the 6581 SID option

dansanderson, 2025-02-23:
**Is your feature request related to a problem? Please describe.**
The MEGA65 core contains two SID implementations labeled "6581" and "8580." The user can choose between these implementations in Configuration, causing all four SIDs to use the selected implementation. I assume we offer this choice because we borrowed an open source implementation that claimed to have two implementations that emulate the subtle differences between (some notion of) the earlier SID 6581 and the later SID 8580.

In practice, we discovered that the implementations were lacking. We have put some effort into improving the accuracy of the 8580, and it would take more effort to similarly improve the 6581. It seems that we're not actually offering a useful choice between two historical recreations due to their inaccuracies.

**Describe the solution you'd like**
Delete the 6581 implementation. Remove the "SID Generation" Configuration option, and reserve the corresponding register for future use. Focus future improvements on the 8580 exclusively.

**Describe alternatives you've considered**
We could invest substantial effort into bringing both implementations up to a degree of accuracy where they not only perform accurately but also recreate the subtle model differences that most people recognize in vintage hardware. This seems unlikely considering it has taken multiple years for community contributors to pull together the skill, interest, and time to produce the fixes we have so far for the 8580, and there is still more work to do just on that implementation.

I may be misunderstanding the current state of the implementation and the possible paths forward. If there is a short path to offering two accurate SID implementations that we just haven't taken yet, maybe the next step is to push in that direction. We have previously considered wholesale replacement of our SIDs with some other SID FPGA project.

**Additional context**
From recent commit messages, I gather the 8580 is in the best shape at the moment. It makes historical sense to focus on the 8580 because that's what the C65 would have used, and I gather that the 8580 is better understood and vintage hardware was produced with tighter tolerances.

If in the future we end up in a position where we can offer accurate 6581 and 8580 options, we can re-add the feature to select between them.

tdeval, 2025-02-23:
FWIW, I'm all for dropping 6581.
It is still available with the C64-core, isn't it ?
Since day 1, I always selected the 8580, and I'm glad it improved over time, 10x bobby_tables.

dansanderson, 2025-04-13:
SteerCo has approved this proposal. This ticket is to track the UI work and VHDL refactoring required to consolidate the current implementation into a single SID, and remove the options from Config.

- [ ] Refactor SID VHDL to only contain the improved 8580
- [ ] Remove SID selection plumbing from the VHDL
- [ ] Remove SID selection UI and storage from Config
- [ ] Remove SID selection UI from User Guide and other relevant documentation