MEGA65/mega65-core #874: EDMA job list cannot reside in Attic RAM

dansanderson, 2025-03-01:
**Test Environment (required)**
You can use MEGAINFO from within FREEZER to obtain this information
(long press RESTORE to enter FREEZER, then press HELP to start MEGAINFO)
 - Platform: MEGA65R6
 - Core Commit: 3c10488
 - ROM Release: 920412

**Describe the bug**
An EDMA job list stored in Attic RAM can be triggered, but it does not succeed. The existence of the megabyte DMA address register D704 implies that this should work.

See the test program source attached, in Acme syntax. The labels `job_addr`, `src_addr`, and `dest_addr` can be adjusted to try different combinations. The program installs an EDMA job to `job_addr` to copy four bytes from `src_addr` to `dest_addr`, sets the source data to AABBCCDD, sets the destination to FFFFFFFF, then executes the EDMA job with its four-byte address. The test is successful if after running the destination memory contains AABBCCDD.

| `job_addr` | `src_addr` | `dest_addr` | Result |
|-|-|-|-|
| $00001600 | $00001700 | $00001800 | Success |
| $00001600 | $08701700 | $00001800 | Success |
| $00001600 | $08701700 | $08701800 | Success |
| $08701600 | $00001700 | $00001800 | Failure |
| $08701600 | $08701700 | $00001800 | Failure |
| $08701600 | $08701700 | $08701800 | Failure |

[testatticdma.asm.txt](https://github.com/user-attachments/files/19037847/testatticdma.asm.txt)

dansanderson, 2025-03-01:
Note that my test program takes the register order into account: bank, megabyte, hi, lo. I'm aware that setting bank also sets megabyte to zero. As far as I can tell, there is a separate issue running a DMA job list located at (for example) 8700000.

MassiveBattlebotsFan, 2025-04-13:
@dansanderson There appears to be a bug in your test. The megabyte register in DMAgic sets address bits 27 to 20, so line 95 should shift by 20, not 24. Modifying it in this manner makes it function perfectly.

dansanderson, 2025-04-17:
Confirmed, thank you! I will clarify this in the documentation.

closed 2025-04-17 by dansanderson